microsoft/vscode-arduino #687: Latest version of Arduino (1.8.7) does not work with extension

kevineger, 2018-11-05:
## Environment Information

System: Ubuntu 18.04
However, I suspect this applies to any and all OSs

## Repro Steps

- Start clean
  - Disable all other extensions
  - Remove existing packages (`~/.arduino15/packages`)
- Install Arduino 1.8.7
- Specify the path in VSCode user settings: `"arduino.path": "/opt/arduino-1.8.7",`
- Attempt to Install a board in _Board Manager_

## Actual

- The _Output_ terminal states "[Done] Installed board package - AZ3166"
- In `~/.arduino15/packages`, the newly _installed_ board is not there

## Expected

- The specified package is installed

## Notes

- Rolling back to 1.8.6, it works.

czgtest, 2018-11-05:
@kevineger Arduino IDE 1.8.7 has some breaking changes, it will install board package and library failed, So it is recommended that you install version 1.8.6

gvarisco, 2018-12-07:
Arduino IDE 1.8.8 has been released and should be hopefully fixing https://github.com/arduino/Arduino/pull/8069. Can you give it a try?

a7hybnj2, 2018-12-08:
I did not have any luck with 1.8.8 or 1.9(beta) with vs-code. I could be overlooking some setup parameter but neither worked.

czgtest, 2018-12-10:
I use Arduino IDE 1.8.8,  install board package and libraries successfully , this issue was fixed on Arduino IDE 1.8.8.

closed 2018-12-11 by czgtest