microsoft/vscode-arduino #690: Unable to get serial monitoring working

dreed47, 2018-11-06:
Im using the latest version of VS Code and the latest version of the extension.  I can upload a sketch fine but when I open serial monitor I get "[Info] Opened the serial port - /dev/cu.Repleo-CH341-00001014" but no further data.  If I then open Arduino IDE and open the Serial monitor there then the monitor starts working in VS Code.  

Any ideas as to what I might look for to see where the problem is?

Thanks.

czgtest, 2018-11-26:
@dreed47  Will this problem occur steadily, or it happen occasionally?  I can not repro on this issue

dreed47, 2018-11-26:
Every time.  The monitor won't work unless I manually open Arduino IDE and then manually open the serial monitor window in Arduino IDE.  Then and only then will I start to see the monitor activity in VZ Code monitor window.

czgtest, 2019-01-21:
@dreed47  Can you have a try on v0.2.25? we add serial port support for Electron 3.0.

chrissearle, 2019-02-19:
I have the exact same symptom here - every time I try it.

```
[Starting] Opening the serial port - /dev/cu.Repleo-CH341-00206214
[Info] Opened the serial port - /dev/cu.Repleo-CH341-00206214
```

This is 0.2.25 talking to a wemos d1 mini - installed arduino app is 1.8.8.

For me also - if I just open the arduino app and open the monitor to the same port - there's all the serial.print output just fine.

adamal, 2019-03-24:
I am experiencing the same thing. 
A further observation, is that when I open the serial monitor, I see a few lines of expected output, then it stops. 
When I start the Arduino IDE monitor, the output starts flowing, but it seems like the serial communication is split between the two more or less randomly, resulting in garbled output. Closing the IDE window, VS Code shows the correct output until the serial connection is restarted.

Arduino IDE: 1.8.9
VS Code: 1.31.1
Arduino extension: 0.2.25

czgtest, 2019-03-26:
@adamal I try to repro this issue on Mac OS 10.14, but the serial monitor always work well , so can you tell me your detail steps?   and which OS?

adamal, 2019-03-26:
@czgtest , strange.
I am on macOS 10.14.3
The arduino code is from example 4 in a book that came with a starter kit. The book can be downloaded here: ([direct download, 8mb](https://bastiaanvanhengel.files.wordpress.com/2016/06/arduino_projects_book.pdf))

kalbergaria, 2019-04-03:
I also experienced this issue. For me, it ended up being that the baud rate in my sketch did not match VSCode's default of 115200. After changing to 115200 in my sketch everything started working great. I believe you could also lower the baud rate to whatever your sketch is using in VSCode's settings if you prefer, but I have not tried this.

czgtest, 2019-04-12:
yes, install the driver for Arduino board ,then select the right port and baud rate, the serial monitor will work well.

WilliamAtek, 2019-07-02:
I have the same problem with my esp32 and ftdi. The problem is caused by the DTR line pulled LOW keeping the esp32 in reset state. You have to let the dtr going back HIGH so that the esp32 can reboot.

desprit, 2019-08-25:
Same problem. I'm using Arduino UNO. Baud rate is 9600 in vscode and in the sketch. Serial port opens but I don't see any output. As soon as I launch serial monitor in Arduino app output starts to appear in the Vscode as well.

Shirsendu-Bairagi, 2019-11-12:
I have a similar problem i am using VS Code 1.40.0 but the serial monitor is not opening as well as the select serial port button is also not working... i just manually added the port. Every time it popups and says Failed to open serial port COM4 due to error: + TypeError: Cannot read property 'close' of null and on the output of serial monitor it says [Starting] Opening the serial port - COM4

rvhani, 2019-11-22:
@Shirsendu-Bairagi ,temporary solution, downgrade vscode to 1.39.2 or older and disable auto update until this issue solved. Work for me. solution from someone on github.

iqapps, 2020-03-11:
Same probleme here. 
Cannot select the serial COM port and also get the error: Cannot get property 'close' on null when trying to open serial monitor.
Downgrading to version 1.42.1 fixes the problem.

hermanius, 2020-05-06:
OSX 10.15.3 - VSCODE 1.44.2 - Arduino extension 0.3.0
For me:
Setting baudrate to 9600 both in sketch and config settings does not work.
Setting baudrate to 115200 both in sketch and config setting works.

NotAnAmbiTurner, 2020-06-05:
Similar issue, but neither baud rate works. Can get output in the IDE serial monitor at either baud rate.

opsaxell, 2020-06-13:
its the 
```
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB
  }
```
part that doesn't work. the actual Serial.print seems to work, if I just skip waiting for the connection. my board is MKR WiFi 1010.

jamesbraza, 2020-06-25:
I am running into this problem today as well on my MacBook (macOS Catalina) + VS Code 1.46.1.

<details>
<summary>settings.json</summary>

My Arduino Due is configured for a baud rate of 9600 bps.

```json
{
    "arduino.defaultBaudRate": 9600
}
```
</details>

<details>
<summary>Versions</summary>

- macOS Catalina: 10.15.5
- VS Code: 1.46.1
- Arduino plug-in: 0.3.1
- Arduino: 1.8.13
</details>

Here's the output.  Sometimes the monitor prints a few lines after `[Info] Opened`, sometimes it does not:

```
[Starting] Opening the serial port - /dev/tty.usbmodem1442401
[Info] Opened the serial port - /dev/tty.usbmodem1442401
```

HEMISixty, 2020-06-30:
This appears to be fixed as of this morning.  I was prompted to update and now Select Serial Port as well as Serial Monitor are both working.

explorer098, 2020-07-03:
I have the same issue with a Adafruit Feather M0 WiFi board. Serial data is only received in the Arduino IDE but not in VS Code. I tried the serial terminal program "hterm", which also did not show any data until I activated the DTR option. Running the exact same code on a Sparkfun Redboard Turbo works as expected and I receive data in VS Code. 
Might the DTR signal cause the issue? How to set it in the VS Code Arduino extension?
VS Code 1.46.1
Arduino Extension 0.3.1
Arduino IDE 1.8.12

DangerousDarlow, 2020-07-31:
> its the
> 
> ```
>   while (!Serial) {
>     ; // wait for serial port to connect. Needed for native USB
>   }
> ```
> 
> part that doesn't work. the actual Serial.print seems to work, if I just skip waiting for the connection. my board is MKR WiFi 1010.

I found this too. The Arduino app must actually send something to the Arduino on connecting where as vscode does not.

Luchiwis, 2020-10-26:
i had the same problem and i solved it by changing the Baud Rate

>ctrl+shift+p >> Arduino: Change Baud Rate

i changed it to 9600 and tested with this code:

```
void setup()
{
  Serial.begin(9600);
}
void loop()
{
  Serial.println("hello\n");
  delay(500);
}
```

OS: Windows 10
VSCode version: 1.48
Arduino extension: v0.3.2
Arduino IDE: 1.8.13

adiazulay, 2020-12-16:
This appears to be resolved, please reopen the issue if you're still have trouble with the serial monitor.

closed 2020-12-16 by adiazulay

sofroniewn, 2021-06-19:
The serial monitor currently doesn't work for me. I just don't see anything. Have tried for both 9600 and 115200 baud rate (matching both what is sent to board and what is in settings, including restarting/ resending everything after any changes).

[image]

[image]

I'm using Mac 10.15.7

VSCode:
Version: 1.57.1
Commit: 507ce72a4466fbb27b715c3722558bb15afa9f48
Date: 2021-06-17T13:28:32.912Z
Electron: 12.0.7
Chrome: 89.0.4389.128
Node.js: 14.16.0
V8: 8.9.255.25-electron.0
OS: Darwin x64 19.6.0

VSCode Arduino extension v0.4.3

Arduino 1.8.15

If I open the Arduino Serial Monitor and set baud rate correct I see output fine there, but it never appears in VSCode

[image]

Any tips would be appreciated, thanks!! I'm new to the vscode-arduino extension (first time user) so might be missing something basic there

Magnatto, 2021-07-12:
I'm having the same issue

ole1986, 2021-08-28:
I could partially get it to work using baudrate 9600 but once I press reset all the "startup/setup" outputs are not displayed

Eggn1n3, 2021-09-01:
I have the same issue; sometimes /dev/USB0 works, sometimes it does not. I am using an Arduino as well.
Version: 1.59.1
Commit: 3866c3553be8b268c8a7f8c0482c0c0177aa8bfa
Date: 2021-08-19T11:54:37.513Z
Electron: 13.1.7
Chrome: 91.0.4472.124
Node.js: 14.16.0
V8: 9.1.269.36-electron.0
OS: Linux x64 5.4.0-81-generic