microsoft/vscode-arduino #700: Support input keystrokes in serial monitor

chyuanmsft, 2018-11-06:


mathertel, 2019-04-13:
There is at least a command "Arduino: Send text to Serial Port" that allows sending some characters.
The only thing that is missing is sending CR or CRLF because enter will send and close the dialog.

Sneezry, 2019-04-15:
Hi @mathertel , there is an option to set line endings. Please follow these steps to set line ending:

1. **Open serial port monitor** (line ending setting only shows when serial port monitor is open)
2. Click line ending setting in status bar
3. Select line ending

[image]

closed 2019-05-29 by stale[bot]

Aravindray, 2020-12-31:
Hi @Sneezry I followed your step but "No line ending" setting is not available in Status bar. So, I can't able to change it. Do you know how to solve it?
[image]

adiazulay, 2021-01-05:
@Aravindray it looks like your serial port is closed in that screen shot. Is the menu option missing when the serial port open too?

vishnumaiea, 2021-06-13:
[image]

It is in fact missing on my status bar. Serial monitor is open as you can see.