microsoft/vscode-arduino #701: Upgrade to VS Code's webview API

mjbvz, 2018-11-06:
Hi, I'm on the VS Code team. I noticed that your extension uses the `vscode.previewHtml` command which is deprecated and which we are actively working to remove: https://github.com/Microsoft/vscode/issues/62630

We've developed a [webview API][docs] that provides a much better developer experience and offers a number of important security and compatibility benefits over `previewHtml`. We cannot fix `previewHtml` without breaking backwards compatibility, and have instead opted to remove it. 

To ensure that your extension continues to work properly in VS Code, please try upgrading to use the new Webview API. You can find documentation on the API usage [here][docs]. Let me know if you have any questions or concerns about this migration

[docs]: https://code.visualstudio.com/docs/extensions/webview

Sneezry, 2018-11-07:
Thanks for the information!

czgtest, 2018-11-22:
This feature is already online on Arduino v0.2.23

closed 2018-11-22 by czgtest