microsoft/vscode-arduino #707: "No Supported board is available" on linux.

Hergeirs, 2018-11-11:
I Have tried following the readme but can't seem to get it too work under linux.

I get this error:
[image]

If i click the "select board type" on the button i get this:
[image]

I'm not sure what is wrong. Would appreciate some help on the matter.
My "arduino.json" content is:

`{
    "board": "arduino:avr:uno",
    "sketch": "sketch_oct30b/sketch_oct30b.ino",
    "port": "/dev/ttyACM0",
    "output": ".build/"
}`

I'm using an Arduino Uno board.

czgtest, 2018-11-12:
@Hergeirs   Can you have a try bellow steps?

1. install Arduino IDE 1.8.6,  don't use Arduino IDE 1.8.7
Note: Arduino IDE 1.8.7 has some breaking changes, as it will install board package and library fails. It is recommended that you install version 1.8.6

2.Open vscode , press F1 and select "Arduino: Board Manager" , install Arduino Uno board package.

3. "select board type" to your install package.

Hergeirs, 2018-11-12:
Thank you. 
Somehow I forgot downgrading the "arduino" package. 
Issue was resolved after installing version 1.8.5 from the repos. 

Will just add this for future reference (and to help googlers).
Using Arch linux you need to add your user to the uucp group to get access to /dev/tty/ACM0.
I.E. run `sudo gpasswd -a <your-user> uucp `

Now I just need to get the includes right in the cpp-preference file :)