microsoft/vscode-arduino #709: [User Feedback] Remember last used baud rate for serial monitor

Sneezry, 2018-11-15:


YakovL, 2020-11-29:
how can I check the actual value of baud rate? A friend of mine showed me that in board config I should see something like

[image]

but what I actually see is

[image]

Since I'm having *major* problems with upload speed (8 minutes for a simple sketch) and I tried to set baud rate to 115200 via ctrl + shift + p → Change Baud Rate, but the problem stays, I'd like to check if baud rate is actually set or not. Having baud rate persistent would be nice, too.

Should I open a new issue? I failed to find anything similar.

adiazulay, 2020-11-30:
@YakovL can you open a new issue, I think this problem is stemming from somewhere else and not the baud rate. Do you experience the same upload speeds in the the Arduino IDE?

gcampbell-msft, 2023-02-16:
Hi @Sneezry, we have recently updated the Arduino extension to depend on a new [Serial Monitor extension](https://marketplace.visualstudio.com/items?itemName=ms-vscode.vscode-serial-monitor). Please update to the newest version of the Arduino extension and let us know what you think and if this issue is still relevant!

closed 2023-04-09 by github-actions[bot]