microsoft/vscode-arduino #718: Unproper closing of Serial Monitor crashes extension

Staud14, 2018-12-04:
Issue Type: <b>Bug</b>

Plug in an Arduino - open serial monitor - press reset button on Arduino - not beeing able to upload a sketch to Arduino anymore

Extension version: 0.2.23
VS Code version: Code 1.29.1 (bc24f98b5f70467bc689abf41cc5550ca637088e, 2018-11-15T19:13:36.375Z)
OS version: Windows_NT x64 10.0.17134

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i7-6700HQ CPU @ 2.60GHz (8 x 2592)|
|GPU Status|2d_canvas: enabled<br>checker_imaging: disabled_off<br>flash_3d: enabled<br>flash_stage3d: enabled<br>flash_stage3d_baseline: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>native_gpu_memory_buffers: disabled_software<br>rasterization: enabled<br>video_decode: enabled<br>video_encode: enabled<br>webgl: enabled<br>webgl2: enabled|
|Memory (System)|15.92GB (11.94GB free)|
|Process Argv||
|Screen Reader|no|
|VM|50%|

</details>

czgtest, 2018-12-05:
@Staud14   Thank you for your feedback ,  we will investigate it.

czgtest, 2018-12-06:
@Staud14 I try to use DevKit AZ3166 and Huzzah Esp8266 device to do bellow steps  ,  but this issue is not repro on .  Can you see what am I missing?

Step:
1.	Open VS Code and install Arduino extension 0.2.23
2.	Open the sample application of   iot-hub-feather-huzzah-client-app
git clone https://github.com/Azure-Samples/iot-hub-feather-huzzah-client-app.git
you can see the tutorial from https://docs.microsoft.com/en-us/azure/iot-hub/iot-hub-arduino-huzzah-esp8266-get-started 
3.	Plug in an Huzzah Esp8266 device to computer.
Auto detect new plugged in board when connect Huzzah Esp8266 to computer in VSCode status 
 
[image]

4.	Run command “Arduino: Open Serial Monitor” to open serial monitor
5.	Press reset button on Huzzah Esp8266 device
6.	then run command “Arduino: upload” to upload a sketh
you can see the Serial Monitor will auto closed when run Arduino upload 
and Arduino upload will success in Arduino panel 
I can not see any crash .

[image]

davystrong, 2019-01-05:
Having this issue too. Interestingly, I was also using a Arduino Nano. When I switched to a Arduino Uno, it didn't happen. Neither are original boards, but the Uno appears as an Uno, whereas the Nano is detected as a "WeMos D1", though I use it as a Nano. Hope this helps.

escaner, 2019-02-04:
I am having this problem in my old laptop (Core 2 Duo) with Linux on an Arduino Micro clone.

When you upload from vscode with the serial monitor open, first the extension closes the monitor, then uploads and finally tries to open the serial monitor again. The problem in my case at that last step is that the serial device /dev/ttyACM0 is not yet ready: the Arduino is either still booting after the upload or the system has not yet updated the filesystem permissions in the /dev/ttyACM0 device (it first appears as crw------ root root and then changes to crw-rw---- root dialout).
Therefore, in the Serial Monitor window I get a "[Starting] Opening the serial port - /dev/ttyACM0" followed by "[Error] Error: Error: permission denied: cannot open /dev/ttyACM0".

Even though the upload was actually successful, in the Arduino upload window, I never get a "[Done] Uploaded the sketch", the message "Arduino uploading..." stays in the status bar and the extension gets "stuck" and cannot be used to upload again until vscode is restarted.

In conclusion, there is a bug in case of error re-opening the Serial Monitor after a successful upload that leaves the upload command unfinished forever.

I would also suggest doing a few tries on the reopen with a little wait between them. When I replace the avrdude executable with my own script adding a sleep 2 (delay 2 seconds) after running avrdude, it just works fine.

I am attaching an animated gif with a successful upload followed by an upload showing the problem with the open serial monitor (I hope it works).

[image]

ziad-akiki, 2019-06-14:
Having this issue when I physically unplug the Arduino without closing the serial port. The extension crashes immediately in this scenario.

gcampbell-msft, 2023-02-16:
@Staudi14 We have recently updated the Arduino extension to depend on our new and improved standalone [Serial Monitor](https://marketplace.visualstudio.com/items?itemName=ms-vscode.vscode-serial-monitor) extension, please upgrade to the latest version to try it out! Let us know if this type of issue is still present.

closed 2023-04-09 by github-actions[bot]