microsoft/vscode-arduino #720: Wrong arduino.commandPath in Doc for MacOS

MarNwk, 2018-12-07:
OS: MacOs High Sierra
Arduino Version: 1.8.7

I installed Visual Code and followed the instructions in the Documentation and used `Contents/MacOS/run-arduino.sh` as path for `arduino.commandPath`, this doesn't work. 
The correct path is `"arduino.commandPath": "Contents/MacOS/Arduino"`

czgtest, 2018-12-07:
@MarNwk ， arduino.commandPath is  option settings,  not necessarily setting, if you don't have customer command line to run ,  you can not set it in VS Code settings.

MarNwk, 2018-12-07:
@czgtest Even if it is optional an optional setting, the value provided for MacOs in the documentation is outdated or wrong.

czgtest, 2018-12-27:
@MarNwk  thank you for your contribution and feedback,  readme already updated

closed 2018-12-27 by czgtest

jimmyff, 2020-09-10:
The text in the settings UI is still incorrect

anarchydude, 2023-01-29:
Can also vouch text is still incorrect as of this post.