microsoft/vscode-arduino #727: intellisense for Arduino Tabs

craiq, 2018-12-16:
Hi,

my project is grown quite big and is devided in many tabs. [Youtube Example](https://www.youtube.com/watch?v=dSAb3LZBHp0)

Now my problem is, that intellisense don't recognize the functions or variables from other tabs and mark them as non declared. I searched a while now and couldn't find an option so intellisense get the information from all files.

Can this please be added? Or does anyone know how to set this up?

System:
Win10
Visual Studio Code 1.30.0
Extensions:
Arduino 0.2.24
C/C++ 0.20.1

czgtest, 2018-12-26:
@craiq ,  you can open your project with vs code and add include path to c_cpp_properties.json

[image]

craiq, 2019-01-08:
Thanks for your reply @czgtest 
just "${workspaceRoot}/**" was missing
the rest already was this way. Still no change.
When I extract parts in another file in the same folder I get undefined errors

disconn3ct, 2019-05-29:
This is still a problem. Please do not close without resolution.

disconn3ct, 2019-06-28:
Must be nice in all of those velocity meetings (or scrum retros or whatever) to be able to say that you have no backlog because its all been closed.. that doesn't make it valid though, and *REALLY* doesn't encourage contributions.

closed 2019-08-02 by stale[bot]

ryancole38, 2020-07-15:
If anyone is still struggling with this, what worked for me was to add a path to an `Arduino.h` file under forcedInclude.

Rovexx, 2021-03-02:
I still encounter this issue on some of my computers. I tried adding paths to all my arduino stuff. also the path to an 'Arduino.h' file. did not work

travipross, 2021-06-18:
Thank you @Rovexx, this solved my issue as well.