microsoft/vscode-arduino #749: non-fatal error on Adafruit Feather M0: cannot open source file "sam.h"

pomplesiegel, 2019-01-10:
When attempting to compile example code or other code for the Adafruit Feather M0, the code successfully compiles. However, this is shown as a "problem" at the top of my project:

{
	"resource": "/Users/michael/Desktop/tempDev/eepromTest/eepromTest.ino",
	"owner": "_generated_diagnostic_collection_name_#0",
	"severity": 8,
	"message": "cannot open source file \"sam.h\" (dependency of \"/Users/michael/Library/Arduino15/packages/adafruit/hardware/samd/1.2.9/cores/arduino/Arduino.h\")",
	"startLineNumber": 4,
	"startColumn": 1,
	"endLineNumber": 4,
	"endColumn": 54
}

I am using the full intellisense feature and have disabled the fallback, as that is the only way I could get syntax highlighting to work. 

Any thoughts? Thank you!

Ivel97, 2019-01-27:
I have the same problem with the Adafruit Metro M0 Express

Ivel97, 2019-01-27:
Fixed by adding:

` "C:\\Users\\------\\AppData\\Local\\Arduino15\\packages\\**"`

Into the includePath in my c_cpp_properties.json file

closed 2019-05-29 by stale[bot]

rickoman, 2020-01-25:
I had the same problem fresh install.  Adding the path recommended by LYWalker fixed the problem