microsoft/vscode-arduino #752: Arduino plugin does not associate with .pde extension

kejxu, 2019-01-16:
ROS is one of the most popular robot framework at this point, and rosserial_arduino is its arduino client support. Most, if not all, of the package's [examples] (https://github.com/ros-drivers/rosserial/tree/melodic-devel/rosserial_arduino/src/ros_lib/examples) use `.pde` extension.

ijash, 2019-03-18:
yes, i need this too..

dooriya, 2019-03-20:
.pde sketch file was supported in Arduino IDE prior to the release of version 1.0.
pre 1.0 versions do not support many new libraries due to significant changes to the Arduino core API. So if you're using old .pde sketch, you can rename it to .ino extension.

kejxu, 2019-03-20:
> if you're using old .pde sketch, you can rename it to .ino extension.

thanks for the explanation! this sounds like the recommended action suggested in [Arduino guide](https://www.arduino.cc/en/Guide/Environment); however, it seems `.ino` and `.pde` are still essentially the same thing.

To clarify the scenario reported, most Arduino examples could be opened and generated from vscode-arduino, like this:
[image]

the generated example defaults to use the `.pde` extension:
[image]

In this case, there would neither be any language support:
[image]

All of these could be fixed by renaming the file's extension to `.ino`; however, in that case the process would not be as stream-lined as it could be. Meantime, this is just one of the cases that is found, there might be dozens more (generated) examples that still use the `.pde` extension (due to lack of maintenance).

Is there a specific reason why `.pde` should not be supported? The extension might change to pop up a message suggesting renaming the extension if the current editor opens a `.pde` file.

closed 2019-03-21 by Sneezry