microsoft/vscode-arduino #763: Source subfolder and linking problem

li3, 2019-01-29:
I'm trying to reorganize my project with subfolders, but I can't seem to get the linker to find the `cpp` sources once they're moved. The extension seems to magically find `cpp` files next to the main `ino` and compile them properly, how do I configure this?

I want to go from this (working structure):

```
project/
├── src/
    ├── app.ino
    ├── some_utility.h
    └── some_utility.cpp
```

To this structure:

```
project/
├── src/
    ├── app.ino
    └── utilities/
         ├── some_utility.h
         └── some_utility.cpp
```

This is my `c_cpp_properties.json` file.

```
{
    "configurations": [
        {
            "name": "Win32",
            "includePath": [
                "C:/Program Files (x86)/Arduino/hardware/arduino/avr/cores/arduino",
                "C:/Program Files (x86)/Arduino/hardware/arduino/avr/variants/standard",
                "C:/Program Files (x86)/Arduino/hardware/tools/avr/avr/include",
                "C:/Program Files (x86)/Arduino/hardware/tools/avr/avr/include/avr",
                "C:/Program Files (x86)/Arduino/hardware/tools/avr/lib/gcc/avr/5.4.0/include",
                "C:/Program Files (x86)/Arduino/hardware/arduino/avr/libraries/**",
                "${workspaceFolder}/src/**",
                "${workspaceRoot}/src/**"
            ],
            "forcedInclude": [
                "C:/Program Files (x86)/Arduino/hardware/arduino/avr/cores/arduino/Arduino.h"
            ],
            "intelliSenseMode": "gcc-x64",
            "compilerPath": "C:/Program Files (x86)/Arduino/hardware/tools/avr/bin/avr-g++.exe",
            "defines": [
                "USBCON",
                "ARDUINO=100",
                "ARDUINO_AVR_NANO",
                "ARDUINO_ARCH_AVR",
                "__AVR_ENHANCED__"
            ],
            "cStandard": "c11",
            "cppStandard": "c++17"
        }
    ],
    "version": 4
}
```

Note, its unclear if `${workspaceFolder}` or `${workspaceRoot}` should be used anymore.

Apologies if this is actually a `vscode-cpptools` question, its kind of tricky to tell where one ends and the other begins sometimes.

li3, 2019-01-29:
I figured it out. To avoid a [Wisdom of the Ages](https://xkcd.com/979/) situation. This has nothing to do with `vscode-arduino`, its a side effect of this change: https://github.com/arduino/arduino-builder/pull/148

Rearranging my folder tree so it looks like:

```
project/
├── src/
│   └── utilities/
│         ├── some_utility.h
│         └── some_utility.cpp
└── app.ino
```

Causes the sub folders to be properly included and compiled.

Morale of the story: your sketch must be in the root of the project, and any other files to compile must within the `src` directory.

closed 2019-01-29 by li3

maikeriva, 2020-02-24:
I recently began working on someone else's code, and it took me an entire day to figure out all those linker errors were due to THIS. Shouldn't this be mentioned in the extension's instructions??

elektronikworkshop, 2020-02-24:
Hi @maikeriva 
This has nothing to do with vscode-arduino. How your project is compiled depends solely on the Arduino-backend which is used by vscode-arduino. vscode-arduino can only operate within the limitations of what the backend provides.
This remains valid as well for the include path search mechanism and library design which people often do not understand.

The rule here is: If something doesn't compile: If it doesn't compile with the Arduino IDE it's a problem with your project. If your project compiles with the Arduino IDE but not with vscode-arduino: You can file an issue here.

elektronikworkshop, 2020-02-24:
BTW: you can help improving the documentation by writing a section for the documentation and file a pull request!
Regards, EW