microsoft/vscode-arduino #764: Arduino Nano with Old Bootloader

kerrsmith, 2019-01-31:
I am trying to set up Arduino for Visual Studio Code so I can program an Arduino Nano. As my Nano is quite old it needs the following board settings in the Arduino IDE:

Board: Arduino Nano
Processor: ATmega328P (old bootloader)

When I search for Arduino Nano in VSCode I do not get the option to choose this old bootloader version, there is just one Nano version - this fails when uploading with the following errors:

avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x61
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x64
...

Is there a way I can change the settings so I am able to program this board?

Thanks in advance for any help.

kerrsmith, 2019-01-31:
After a bit of searching I saw that a good way to fix this issue was to update the bootloader on my Arduino Nano.

I used a USBasp module and the Arduino IDE to do this, once it was upgraded I chose 'Arduino Nano' from the 'Arduino: Board Manager' setting in VSCode and my program compiled and uploaded as expected.

I am really pleased this now works as using VSCode to program is really a great help as it has so many helpful features.

Thanks for making this extension, it is really appreciated.

fakefred, 2019-03-25:
For those whose config panel does not work, you can edit .vscode/arduino.json and change "configuration" to "cpu=atmega328old".

mattjanssen, 2021-08-01:
Just for reference, here are the two places I had to click in order to find the **Arduino Board Configuration** screen:
[image]
[image]

tomotvos, 2022-03-19:
+1000 for this!