microsoft/vscode-arduino #765: How do you enable error highlighting

kerrsmith, 2019-01-31:
I have installed the extension and it works as expected, I can compile and upload programs to my Arduino nano.

The issue I am currently seeing is there is no error checking as I enter new code. If I miss out a comma or spell a function name wrong I do not see the usual red line under the error.

I have not made any changes to the default install except add "output": "../build" to the arduino.json file.

Do I need to do anything to enable error checking?

My current set up is as shown below:

c_cpp_properties.json is:

{
    "configurations": [
        {
            "name": "Win32",
            "includePath": [
                "C:\\Users\\username\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\**",
                "C:\\Users\\username\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.6.23\\**",
                "C:\\Users\\username\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\**",
                "C:\\Users\\username\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.0-beta3\\**"
            ],
            "forcedInclude": [
                "C:\\Users\\username\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.6.23\\cores\\arduino\\Arduino.h"
            ],
            "intelliSenseMode": "msvc-x64"
        }
    ],
    "version": 4
}

VScode details:

Version: 1.31.0-insider (user setup)
Commit: 7f136b44cff0cd3486ac88b43598b8c72b025d16
Date: 2019-01-31T08:03:26.905Z
Electron: 3.1.2
Chrome: 66.0.3359.181
Node.js: 10.2.0
V8: 6.6.346.32
OS: Windows_NT x64 6.1.7601

kerrsmith, 2019-02-01:
After quite a while experimenting it appears the errors do work on a new almost empty project but as soon as you include headers to it that is when the errors seem to stop showing.

It appears that if there are any problems with the program locating headers, or issues in the headers then this stops all further errors being shown. When you get the green line under a header this highlights it is having issues but it appears even if the green line does go when you add the header location to the c_cpp_properties.json file it can still have problems.

Currently I have no green lines under the headers but I do not get errors showing.

Also, I need to set C_Cpp:Intelli Sense Engine to ''Tag Parser" in the C/++ settings or I get lots of the following errors when hovering functions (and 'Go To definition' does not work):

Quick info operation failed: Error: Failed to find translation unit. Try opening a source file that includes this header, then re-open this header.

kerrsmith, 2019-02-01:
After much for experimenting I think the issue for this is actually with ms-vscode.cpptools extension as it is the part of code that deals with all the errors and intellisense details. I think vscode-arduino is working fine as I can compile and upload successfully.

closed 2019-02-01 by kerrsmith

kerrsmith, 2019-02-02:
I have now got error highlighting working with my Arduino code, here are the steps I used to get it working:

Create a new project folder
Launch VScode and open this new folder
Use CTRL+Shift+P and choose 'Arduino: Initialize'
Name your .ino file
Choose the relevant board (I choose 'Arduino Nano')

Next open this new .ino file in the official Arduino IDE and go to File -> Preferences
Tick the 'Show verbose output during compilation' and then click OK.
With the .ino file open and the correct board chosen select 'verify', this is the 'tick' icon.
Copy the text from the bottom of the window into a text editor and look for the line that says 'Detecting libraries used...'
Just under this look at the line similar to the following:

`C:\\Users\\YOUR-USER-NAME\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\5.4.0-atmel3.6.1-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10808 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR`

Copy the part from `-c` to `-DARDUINO_ARCH_AVR`

Go back to VScode and open 'c_cpp_properties.json' from the .vscode folder

Replace the contents of this file with the code shown below, use the part you copied above in the 'compilerPath' line:

```JSON
{
    "configurations": [
        {
            "name": "Win32",
            "includePath": [
                "C:\\Program Files (x86)\\Arduino\\tools\\**",
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\**",
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\**",
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\**",
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\bootloaders\\**",
                "C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr\\avr\\include"
            ],
            "forcedInclude": [
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino\\Arduino.h"
            ],
            "intelliSenseMode": "gcc-x64",
            "compilerPath": "\"C:/Program Files (x86)/Arduino/hardware/tools/avr/bin/avr-g++.exe\" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10808 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR",
            "cStandard": "c11",
            "cppStandard": "c++17"
        }
    ],
    "version": 4
}
```
 (thank you to https://github.com/Microsoft/vscode-arduino/issues/684#issue-377193346 for this whole section of code)

Save the file and wait until the small red icon and three small disk icons disapear from the bottom right hand corner of the window - this allows VScode time to index all the files. Do not do anything else whilst this indexing is taking place as it seems to stop it completing correctly - just sit and wait for it to finish.

Now you can go back to your .ino file and start coding and the intellisense and error highlighting will now work.

If you have additional libraries just add a link to them in the 'includePath' part in c_cpp_properties.json.

cmwhittaker, 2020-07-09:
You MUST delete this line
  "C_Cpp.intelliSenseEngine": "Tag Parser"
if you have put it in the settings.json (Cntrl + Shift + P search "Settings.json")