microsoft/vscode-arduino #766: It works just as a plain text editor

Mark-81, 2019-02-01:
I've installed the extension and it seems to work: it build and download application to my Arduino boards.
I switched to Visual Code Studio to benefit of a more serious editor than the default Arduino IDE.

Unfortunately a lot of features don't work:

- compiler errors and warnings don't show up in the "Problems" tab
- I see them in the "Output" tab but I cannot navigate to the related line, nor I see their highlights in the code
- "go to definition" command works, but "Find/Peek all references" doesn't
- every time I "verify" the code it says: "compiling option changed: rebuild all"

It's my setup that doesn't work properly?

czgtest, 2019-02-13:
@Mark-81 

- compiler errors and warnings show up in the "Output" tab instead of "problems" tab
- it is not support to navigate to the related line and highlights , the same request #770 
- yes
- if you don't want to rebuild all each time, you can add "output" path in arduino.json
[image]

Mark-81, 2019-02-21:
@czgtest 

> if you don't want to rebuild all each time, you can add "output" path in arduino.json

I added the `output` path in arduino.json but it still says "compiling option changed: rebuild all" every time

adiazulay, 2021-03-22:
IntelliSense auto-generation has been added in v0.4.0. If you're having any problems with IntelliSense please open a new issue.

closed 2021-03-22 by adiazulay