microsoft/vscode-arduino #768: Support Arduino IDE 1.8.8

mads256c, 2019-02-02:
Verifying the sketch fails with java nullpointerException

```
java.lang.NullPointerException
	at processing.app.BaseNoGui.onBoardOrPortChange(BaseNoGui.java:679)
	at processing.app.Base.onBoardOrPortChange(Base.java:1313)
	at processing.app.Base.<init>(Base.java:276)
	at processing.app.Base.main(Base.java:145)
```

czgtest, 2019-02-13:
@mads256c  Arduino extension v0.2.25 already support Arduino IDE 1.8.8 now,  verify and upload can work well on my machine with vscode 1.30.2.

do you install board package or library before verify the sketch ?  

can you share more detail about how to repro this issue?

mads256c, 2019-02-24:
I downloaded the arduino ide through the arch linux package system. Which uses custom board definitions in another package. I simply had to open Arduino IDE then go to board manager and download the standard board definitions and it worked.

closed 2019-02-24 by mads256c