microsoft/vscode-arduino #770: Problem Matcher

alexeystn, 2019-02-05:
I have initialized my Arduino project in VS code.
Verify/Upload/Serial and other functions works fine there.
The only thing that does not work is matching build errors and warnings with code. For instance, when I see an error message in the output window like this:
`myProject:120: error: 'myFunction' was not declared in this scope`
I cannot click this message and jump to line 120. Instead I must find this line manually.
Is there now a possibility to configure a problem matcher for Arduino in VS code?

Marsfan, 2021-01-01:
I have found that the the `$gcc` problem matcher does an ok job at matching problems from a Arduino Custom Build Task (as long as you tell it that it uses absolute paths), but it gets confused when the output from Arduino provides a multi-line chain of files to get the correct location of the error/warning. 

This is not leveraging this extension at all, but I believe that it can probably be adapted for use with this extension. 

```json
            "label": "Verify Active File",
            "type": "shell",
            "command": "C:/Program Files (x86)/Arduino/arduino_debug.exe",
            "args": [
                "--preserve-temp-files",
                "--verbose",
                "--board",
                "SparkFun:apollo3:sfe_artemis_atp:svl_baud=921600,loader=option_svl",
                "--verify",
                "${file}",
                "--pref",
                "build.path=${workspaceFolder}/.build",
                "--pref",
                "preproc.color_datatype=true",
                "--pref",
                "preproc.web_colors=true"
            ],
        
            "problemMatcher": {
                "base": "$gcc",
                "fileLocation": [
                    "absolute"
                ]
            },
```

Marsfan, 2021-01-01:
Well it was working, but apparently changed its mind, as is custom with anything I try to do it seems. The Arduino IDE suddenly decided to start using multi line error messages, I am working on implementing a functional problem matcher, and will post it here once I get it working.