microsoft/vscode-arduino #772: How to fix red squiggles under constants like D2 (upload works fine)

DanTup, 2019-02-08:
I've got everything working, except the constants for pins don't seem to be defined:

[image]

The upload works fine, and I have no issues in the standard Arduino IDE. I think these come from a file like:

https://github.com/esp8266/Arduino/blob/master/variants/d1/pins_arduino.h

Though ideally I'd like it to be based on the board I have selected automatically (since I switch between different boards).

DanTup, 2019-02-08:
I seem to have fixed it by changing:

> "/Users/danny/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/**",

to:

> "/Users/danny/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/variants/d1/**",

This doesn't seem ideal because I'd have to change it all the time - is there a way to tie these to the board selection?

adiazulay, 2021-03-22:
IntelliSense auto-generation has been added in v0.4.0. If you're having any problems with IntelliSense please open a new issue.

closed 2021-03-22 by adiazulay