microsoft/vscode-arduino #774: Failed to open serial port /dev/ttyUSB0 due to error: + TypeError: Cannot read property 'close' of null

meier99, 2019-02-14:
Issue Type: <b>Bug</b>

Always happens on 'Arduino: Open Serial Monitor'.

Upload of sketch (Arduino: Upload) to board via /dev/ttyUSB0 works.
Serial Monitor started via Arduino application also works fine.

Extension version: 0.2.25
VS Code version: Code 1.31.0 (7c66f58312b48ed8ca4e387ebd9ffe9605332caa, 2019-02-05T22:32:48.719Z)
OS version: Linux ia32 4.15.0-45-generic

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i3-2330M CPU @ 2.20GHz (4 x 1167)|
|GPU Status|2d_canvas: enabled<br>checker_imaging: disabled_off<br>flash_3d: enabled<br>flash_stage3d: enabled<br>flash_stage3d_baseline: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>native_gpu_memory_buffers: disabled_software<br>rasterization: disabled_software<br>surface_synchronization: enabled_on<br>video_decode: unavailable_off<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|0, 0, 0|
|Memory (System)|3.89GB (0.60GB free)|
|Process Argv|--unity-launch|
|Screen Reader|no|
|VM|0%|

</details>

czgtest, 2019-02-19:
I will investigate with it.

edalongeville, 2019-04-22:
Hello,

I am facing the same problem on Manjaro 18, running extension version 0.2.25, VSCode 1.33.1 and arduino 1.8.9.

"Arduino: Upload" works perfectly, but opening the serial port causes the following error:

> Failed to open serial port /dev/ttyUSB0 due to error: + TypeError: Cannot read property 'close' of null

This is not a permission issue. I put my ttyUSB0 ad chmod 777 to troubleshoot, the problem is still here.
Using the same software versions, it works on my Ubuntu 18.04 Laptop.

Any logfile you'd like me to check?

pg1992, 2019-04-28:
I'm having the same issue. When I try to run <kbd>F1</kbd> `Arduino: Open Serial Monitor` I get the warning
```
Failed to open serial port /dev/ttyUSB0 due to error: + TypeError: Cannot read property 'close' of null
```
Verify and upload run without any problems, and I can access the serial port running `screen /dev/ttyUSB0 115200` on my terminal emulator.

If I run my project with `code --verbose` I get this in the output:
```
...
[17693:0428/103656.112433:INFO:CONSOLE(3214)] "[[object Object]]Cannot read property 'settings' of null", source: file:///usr/lib/code/out/vs/workbench/workbench.main.js (3214)
[17693:0428/103656.112481:INFO:CONSOLE(3214)] "TypeError: Cannot read property 'settings' of null
	at SerialMonitor.initialize (/home/espectro/.vscode-oss/extensions/vsciot-vscode.vscode-arduino-0.2.25/out/src/serialmonitor/serialMonitor.js:42:69)
	at constructor.dc.onDidChange (/home/espectro/.vscode-oss/extensions/vsciot-vscode.vscode-arduino-0.2.25/out/src/serialmonitor/serialMonitor.js:26:26)
	at u.fire (/usr/lib/code/out/vs/workbench/services/extensions/node/extensionHostProcess.js:44:708)
	at loadContext.vscode.workspace.findFiles.then (/home/espectro/.vscode-oss/extensions/vsciot-vscode.vscode-arduino-0.2.25/out/src/deviceContext.js:76:39)", source: file:///usr/lib/code/out/vs/workbench/workbench.main.js (3214)
...
```
 
---
I'm running it on
- Arch Linux
- `code --list-extensions --show versions`:
  ```shell
  dbaeumer.vscode-eslint@1.8.2
  donjayamanne.jquerysnippets@0.0.1
  dracula-theme.theme-dracula@2.17.0
  eamodio.gitlens@9.6.3
  esbenp.prettier-vscode@1.9.0
  mrmlnc.vscode-autoprefixer@2.2.0
  ms-python.python@2019.4.11987
  ms-vscode.cpptools@0.22.1
  PKief.material-icon-theme@3.7.0
  vsciot-vscode.vscode-arduino@0.2.25
  ```
- `code --version`:
  ```shell
  1.33.1
  51b0b28134d51361cf996d2f0a1c698247aeabd8
  x64
  ```
- `uname -a`:
  ```shell
  Linux arch-avell 5.0.8-arch1-1-ARCH #1 SMP PREEMPT Wed Apr 17 14:56:15 UTC 2019 x86_64 GNU/Linux
  ```

hywhuangyuwei, 2019-05-10:
I have this problem too.
I am using Manjaro with Arduino IDE 1.8.9 and _Verify & Upload_ also works perfectly.
Note: I noticed that when I click the _select serial port_ there is no response and I had to write that to the arduino.json manually.

Moonlight63, 2019-06-10:
@hywhuangyuwei Same problem here with no response clicking select serial port. In my case, uploading from arduino also did not work. I had to use `sudo chmod a+rw /dev/ttyACM0` to upload anything.

aboulfad, 2019-07-03:
Me too, just happened after VSCode updated itself to the latest June 2019 version ...

dbaeumer, 2019-07-04:
Regarding @aboulfad comment above: See this comment here for why this fails right now: https://github.com/microsoft/vscode/issues/76538#issuecomment-508385062

ArthurMa1978, 2019-07-09:
Fixed since Arduino extension v0.2.27, thanks @meier99 @edalongeville @pg1992 @hywhuangyuwei @Moonlight63 @aboulfad @dbaeumer .

edalongeville, 2019-07-10:
Thanks a lot @ArthurMa1978 !

PeterWone, 2019-09-22:
It's not fixed in 0.2.27 for me, in fact it only APPEARED in this version ON WINDOWS.

DoganM95, 2019-09-24:
the above error **"Failed to open serial port COM3 due to error: + TypeError: Cannot read property 'close' of null"** still persists for me.

VSCode Version: 1.39.0 (insiders)
Arduino Extension Version: 0.2.27

AdrienTD, 2019-11-10:
I seem to now get this issue too after updating to the latest version of VSCode (v1.40.0). I can still compile and upload programs to my ESP32, but can't open the serial monitor anymore. Before the update I never had this problem. Opening the serial monitor still works in the Arduino IDE fortunately.

janperner-ubnt, 2019-11-10:
Exactly the same issue here. It has stopped working after an upgrade to v1.40.0.

bukodi, 2019-11-10:
#MeToo :)

After upgrade VSCode to v1.40.0, serial monitor causes "TypeError: Cannot read property 'close' of null"  (Ubuntu 18.04.3 LTS)

bluesnotes, 2019-11-11:
Me too 
VSCODE 1.40.0
Arduino extension version 0.2.27
Arduino 1.8.10

since upgrade VSCode to 1.40.0

UKSFM99, 2019-11-11:
Me too since upgrading to vscode 1.4.0

luke-shields, 2019-11-11:
Same issue here, also not able to select serial port as option

serioushare, 2019-11-12:
Got the same issue, but I'm mostly using the arduino devtools serial monitor since it's 2 way.

@luke-shields:  You can select the port by adding `"port": "COM3",` to the projects arduino.json file.

xdubx, 2019-11-13:
Same issue on vsc 1.4.0 on linux. Serial port selecting with a waring in verbose mode. But no error message.
Flashing works, if you set the port manually.

PeterWone, 2019-11-13:
The problem is well understood and has a straightforward solution. This extension uses native node modules which breaks whenever the node version of VS Code changes. The solution is to update the extension to use the new version of node, but this just breaks again later for the same reason &ndash; which is why this issue just won't die.

A permanent solution suggested by the VS Code team is to rewrite the unsupported code to use N-API, which is what they're adopting (VS Code also needs to call native modules).

bluesnotes, 2019-11-15:
> The serial port issue has been fixed in the latest Arduino extension
> release (0.2.28).
> Please install the latest Arduino 0.2.28 from marketplace
> https://marketplace.visualstudio.com/items?itemName=vsciot-vscode.vscode-arduino
> to verify if this is resolved on your environment.

Perfect! Thank's.

0xStarcat, 2019-12-02:
I just installed 0.2.28 and I still have the issue on linux. (arduino 2:1.0.5)

PeterWone, 2019-12-02:
@0xStarcat  What version of VS Code do you have?

DoganM95, 2020-01-25:
[image]

VSCode 1.42.0 Insiders
Arduino (VScode extension) 0.2.29

0xStarcat, 2020-01-25:
> @0xStarcat
> 
> * What version of VS Code do you have?
> * What distro and build of Linux?
> 
> I won't insult your intelligence by asking whether you've tried rebooting. You shouldn't need to. But if you haven't, now would be a good time to rule out orphaned locks or hung processes as contributing factors.

Sorry for late reply. I can't provide more information because I don't remember what I was doing to cause it (doh!). Consider it resolved for me.

DoganM95, 2020-02-27:
It has been a year, since this issue was opened and is still unsolved. 
Any progress so far on this problem? This issue really renders the vscode-arduino extension useless.

hellyzh, 2020-02-28:
@DoganM95 , I believe the issue has been fixed in the latest version. If you still has issue, please check your OS configuration, if you use linux, please check USB config according to [Guid ](https://www.arduino.cc/en/Guide/Linux):
`ls -l /dev/ttyACM*`
`sudo usermod -a -G dialout <username>`

hellyzh, 2020-03-06:
@DoganM95 , I created a new issue #980 to track this. 
Please try to downgrade to VSCode 1.42 to work-around.

DoganM95, 2020-03-06:
@hellyzh thanks for keeping track of this! 
The Serial Port issue still persists when downgrading to vscode 1.42

Kowiste, 2020-03-10:
Same problem it was working perfectly and suddenly
[image]

hellyzh, 2020-03-10:
@Kowiste, it is due to VSCode auto upgraded to version 1.43, currently please use [VSCode 1.42 ](https://update.code.visualstudio.com/1.42.1/win32-x64/stable) as a workaround.

rrakso, 2020-03-10:
[image]
The same! :((

cooperbaker, 2020-03-10:
Same here :( 

downgrade is a poor solution

Xring-git, 2020-03-10:
The same here!

StratosMylonas, 2020-03-11:
VSCode v1.43.0
Arduino v0.2.29
Windows 10 x64
Same here...

Christian1984, 2020-03-11:
I'm having the same issue here:

Mac OSX 10.14.6
Arduino IDE 1.8.11
VSCode 1.43.0
Arduino-Extension 0.2.29

I also had to set my port manually through arduino.json since the port selection did not work otherwise (see https://github.com/microsoft/vscode-arduino/issues/922).

duckbill70, 2020-03-12:
Hi, same issue for me:

Mac OSX 10.15.2
VSCode 1.43.0
Arduino-Extension 0.2.29

bzeeman, 2020-03-12:
Same issue. Honestly Microsoft, your default forced updates are almost always a problem for developers. You should rethink your choice own that when it comes to applications designed for WORK.

MacOS 10.15.3
VSCode 1.43.0
Arduino-Extension 0.2.29

collins-kariuki, 2020-03-12:
was having the same issue;thought I had broken something :/

hussam0O, 2020-03-12:
I have the same issue.
is there are any quick solution ?

collins-kariuki, 2020-03-12:
> I have the same issue.
> is there are any quick solution ?

Yes just revert to v 1.42

PeterWone, 2020-03-12:
@bzeeman The problem is well understood and the workaround is public knowledge. The way to solve it is also clear. If you think it's a priority item, why don't you fork the project, rewrite it to use N-API and submit a PR?

dilin-MS, 2020-03-23:
Duplicate with #980

closed 2020-03-23 by dilin-MS

apicquot, 2020-03-24:
Had the issue with
VSCode v1.43.0
Arduino v0.2.29
Windows 10 x64

Works again after downgrading to
VSCode v1.42.0
Arduino v0.2.29
Windows 10 x64

I simply downloaded https://update.code.visualstudio.com/1.42.1/win32-x64/stable and clicked for the installed

HEMISixty, 2020-06-29:
Hello folks, I'm running into the same problem.  Unfortunately, "Arduino: Select Serial Port" doesn't do anything.  I can manually update my arduino.json file to specify the port for my WiFi board as a workaround.  I really don't want to downgrade my VSCode installation.  But, I'll give that a shot down the road if this isn't an easy fix.

VSCode v1.47.0-insiders
Arduino IDE is 1.8.13
Arduino Extension is 0.3.1
Windows 10 Pro 19042.330

HEMISixty, 2020-06-30:
> Hello folks, I'm running into the same problem. Unfortunately, "Arduino: Select Serial Port" doesn't do anything. I can manually update my arduino.json file to specify the port for my WiFi board as [...]

Update, as of this morning, the problem has been resolved.  When I fired up VSC, I didn't much attention but something updated and now the COMM Port selection and Serial Monitor are both working... I've checked my versions and nothing reflects an update.  So, I don't know what updated.  But, my serial connectivity is working...

HEMISixty, 2020-07-10:
Well, as of this morning, apparently a VSC update has broken the extension.

VSCode v1.48.0-insiders
Arduino IDE is 1.8.13
Arduino Extension is 0.3.1
Windows 10 Pro 19042.330

This is the current error message:  

Failed to open serial port COM5 due to error: + Error: \\?\c:\Users\Sixty\.vscode-insiders\extensions\vsciot-vscode.vscode-arduino-0.3.1\out\node_modules\usb-detection\build\Release\detection.node is not a valid Win32 application.
\\?\c:\Users\Sixty\.vscode-insiders\extensions\vsciot-vscode.vscode-arduino-0.3.1\out\node_modules\usb-detection\build\Release\detection.node

I backed VSC down to 1.47.0-insiders and that didn't resolve the problem.  I've uninstalled the extension to reinstall it and that doesn't resolve the problem.

Code still compiles and deploys.  I just can't use serial monitor.  Any help would be greatly appreciated...and, I've disabled automatic VSC updates from here on out...