microsoft/vscode-arduino #776: cannot open source file "omp.h" (dependency of "ESP8266WiFi.h"

Cictani, 2019-02-16:
I get the following error when trying to run my code which works fine if using arduino ide:

[image]

andynabokus, 2019-02-26:
Hi! You need configure you include path in the c_cpp_properties.json file.
raw "/opt/arduino_1_8/**" - includes any files in the arduino directory.
{
    "configurations": [
        {
            "name": "Linux",
            "includePath": [ 
                "${workspaceFolder}/**",
                "/opt/arduino_1_8/**"
            ],
            "defines": [],
            "compilerPath": "/usr/bin/clang++-4.0",
            "cStandard": "c11",
            "cppStandard": "c++17",
            "intelliSenseMode": "clang-x64",
            "forcedInclude": []
        }
    ],
    "version": 4
}

pmckinney1971, 2019-04-22:
The above reply is for an entry for the include path on a Linux machine ("/opt/arduino_1_8/**"), but don't know what the equivalent would be on a Windows machine?   ... I have exactly the same error as Cictani above.

Current include path is (some of which looks redundant to me):
```
"includePath": [
                "C:\\Users\\Leonie\\Documents\\Arduino\\hardware\\esp8266com\\esp8266\\libraries\\ESP8266WiFi\\src",
                "C:\\Users\\Leonie\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\**",
                "C:\\Users\\Leonie\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.0\\**",
                "C:\\Users\\Leonie\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.0\\libraries\\**",
                "C:\\Users\\Leonie\\AppData\\Local\\Arduino15\\**",
                "C:\\Users\\Leonie\\Documents\\Arduino\\hardware\\esp8266com\\esp8266\\**",
                "C:\\Users\\Leonie\\Documents\\Arduino\\libraries\\**",
                "${workspaceFolder}/**"
```

oparex, 2019-11-20:
I have the same problem on a mac. I can't find the location of omp.h, I tried to find it with find in terminal, no success. It would help to know Arduino IDE's include path. Google searches on this only explain what this is, not where...

stoonoon, 2020-01-12:
Same thing there.... I couldn't find it anywhere on my win 10 machine. A bit of googling suggested that it's part of OpenMP, which was included in gcc, but not in the xtensa-lx106-elf-gcc folder included in the esp8266 package. I installed gcc via WSL, and then found omp.h under %LOCALAPPDATA%\lxss\rootfs\usr\lib\gcc\x86_64-linux-gnu\5\include 
I copied it to a new folder under Arduino\libraries\ and that error has gone.
It feels like a horrific bodge, but it's let me move on for now, so I think I can live with it....

itdir, 2020-06-08:
Same issue...

macOS 10.14.6
VSC 1.45.1
Arduino 1.8.12

I've tried many variations of the Include path, but now I'm back to
${workspaceFolder}/**
${HOME}/Documents/Arduino/libraries/**

luisrival, 2020-07-15:
Someone know how to fix it for windows? I have the same issue

mitchh, 2020-07-17:
`ESP8266WifiType.h` (included by `ESP8266WiFi.h`) is finding the wrong `queue.h`, which is then trying to find some OpenMP headers.  I fixed it on my Mac by putting the sdk include specifically before the the broad `**` include paths. Here is a (redacted; everything in angle brackets has been removed/replaced. Also your version numbers may be different) snippet of my `c_cpp_properties.json`:

```json
      "includePath": [
        <more stuff>
        "<my home dir>/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.2/tools/sdk/include",
        "<my home dir>/Library/Arduino15/packages/esp8266/tools/**",
        "<my home dir>/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.2/**"
      ],
```

The `tools/**` line matched:

```sh
$ find ~/Library/Arduino15/packages/esp8266/tools -iname queue.h 
<my home dir>/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/xtensa-lx106-elf/include/c++/4.8.2/parallel/queue.h
<my home dir>/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/xtensa-lx106-elf/include/sys/queue.h
```

...and the one of those it used tried to pull in the OpenMP stuff. 

The `hardware/esp8266/2.7.2/**` one matched the right one (among other things):

```sh
$ find ~/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.2 -iname queue.h
<my home dir>/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.2/tools/sdk/libc/xtensa-lx106-elf/include/sys/queue.h
<my home dir>/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.2/tools/sdk/include/queue.h
<my home dir>/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.2/tests/host/common/queue.h
```

but a) was already too late after the `tools/**` line and b) still gave too many answers.

I think the `**` are scary resource-wasting monsters, but I don't have (haven't looked hard enough for) a hassle free alternate solution (yet?)

MohammedNoureldin, 2020-11-06:
OMG I am spending time in solving such issues longer than coding... I have the exact same problem.

MohammedNoureldin, 2020-11-07:
OK, it turned out that I could fix it (or work around it), by specifying the path of the compiler, just like this:

```
"compilerPath": "C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr\\bin\\avr-gcc.exe",
"intelliSenseMode": "gcc-x64",
```

jevoswang, 2020-12-05:
Create an empty omp.h by yourself, and there is a relevant path.

adiazulay, 2021-03-22:
IntelliSense auto-generation has been added in v0.4.0. If you're having any problems with IntelliSense please open a new issue.

closed 2021-03-22 by adiazulay