microsoft/vscode-arduino #784: Output Panel stops auto-scroll

Drayak335, 2019-03-08:
The Output Panel suddenly stops auto-scrolling for "verify" or "upload", I need to manually scroll.  I don't know how to replicate it, but I have experienced several times. 

The only way I have found so far to make it auto-scroll again is to completely close the application and launch again.

dilin-MS, 2019-03-11:
Hi @Drayak335 , can you please provide a snapshot so we can better the situation you met? Many thanks!

Drayak335, 2019-03-11:
@dilin-MS  by snapshot, do you mean a screenshot? If not how do I get the snapshot?

Below a screenshot, if you can see in the right corner, when I hit verify I see the message "[Starting] Verfy sketch..."  and then it starts throwing all the messages, but the window doesnt auto-scroll, I need to  manually scroll to read the messages.

[image]

Sneezry, 2019-03-12:
Hi @Drayak335 , click the small lock, and you will get auto scoll back.

[image]

Drayak335, 2019-03-12:
> Hi @Drayak335 , click the small lock, and you will get auto scoll back.
> 
> [image]

Drayak335, 2019-03-12:
> Hi @Drayak335 , click the small lock, and you will get auto scoll back.
> 
> [image]

Is there any keyboard shortcut to toggle this? I am trying to understand how it may have changed.

closed 2019-03-12 by Drayak335

Sneezry, 2019-03-12:
I don't know the keyboard shortcut for the lock, and I cannot find it in shortcut settings either. So I think the anwser is no.

Drayak335, 2019-03-22:
Just as a follow up, I noticed that at some point something changes the state of this setting. I have not been able to figure out what is triggering the change?

cocohub, 2020-03-30:
I have the same issue. Even though I click the lock icon, it stops auto scrolling. Any tips?