microsoft/vscode-arduino #785: Extension causes high cpu load

matthewburton71, 2019-03-10:
- Issue Type: `Performance`
- Extension Name: `vscode-arduino`
- Extension Version: `0.2.25`
- OS Version: `Darwin x64 18.2.0`
- VSCode version: `1.32.1`

:warning: Make sure to **attach** this file from your *home*-directory: `/Users/matthew/vsciot-vscode.vscode-arduino-unresponsive.cpuprofile.txt` :warning:

Find more details here: https://github.com/Microsoft/vscode/wiki/Explain:-extension-causes-high-cpu-load

dilin-MS, 2019-03-11:
Hi @matthewburton71 , do you mean you encountered high cpu load due to vscode-arduino? Can you provide more description concerning your situation?

zunhu, 2019-08-26:
issue fixed, please install latest vscode\ arduino-extension\ arduino and try again.

closed 2019-08-26 by zunhu