microsoft/vscode-arduino #786: New feature: Compile to *.bin

Pantastisch, 2019-03-13:
The Arduino IDE suffers an option to build a sketch as *.bin file.

Adding this feature to VS Code Arduino Plugin would be impressive.
This feature is mostly used, to upload code wireless to an ESP.

You can find this option on the Arduino IDE on Sketch -> Export compiled Binary.
It will be placed in the Sketch folder.

Sneezry, 2019-03-14:
Hi @Pantastisch , thanks for your feedback. In fact, the extension talks with Arduino IDE with a cli interface, but not the GUI. Unfortunately, the cli interface doesn't have the full function of the GUI version, and that is the limit of this extension. A full support list of the cli interface can be found here: https://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc

Also, that is the reason why some functions we can find in Arduino IDE is missing in the extension. As Arduino IDE is beyond from our control, some work is not as easy as it looks like.

Your suggestion is valuable, and we will consider it serious. However, please understand that is not as easy as it looks.

Pantastisch, 2019-03-19:
Alright! Thanks for the reply @Sneezry !

elgerg, 2019-05-11:
What about the following:

Compile a sketch, put the build results in the build directory an re-use any previous build results in that directory.

arduino --pref build.path=/path/to/sketch/build --verify /path/to/sketch/sketch.ino

Could you not do a verify and then copy the bin to the sketch folder?

MickGyver, 2020-02-27:
Any news on this? I was hoping to switch to VS Code for Ardiuino development, but this issue is a game breaker.

elektronikworkshop, 2020-02-27:
Hi @MickGyver 
as @elgerg outlined you can do the following: Specify an output path (what you want anyways to speed up your builds) and then you could fetch the bin from that output path.

A possible feature which I would propose is a post build command (analog to the pre build command) which allows for scripts/commands to be run after a build. By means of such a command you could easily move files back and forth and even upload, archive, ... to your liking.

This would be simple to implement but I'm currently busy implementing #438 and currently there's not much response from the repository maintainers.

Regards,  
EW

MickGyver, 2020-02-27:
> as @elgerg outlined you can do the following: Specify an output path (what you want anyways to speed up your builds) and then you could fetch the bin from that output path.
Thanks! I missed that, this does just what I want. :)

> A possible feature which I would propose is a post build command (analog to the pre build command) which allows for scripts/commands to be run after a build. By means of such a command you could easily move files back and forth and even upload, archive, ... to your liking.
That would be great! I run a batch file now manually post build.

elektronikworkshop, 2020-02-27:
Great!
I'll let you know if I find I minute to implement this. It will then be featured in one of my alpha releases I usually announce alphas in the issue above or you can follow the progress [here](https://github.com/elektronikworkshop/vscode-arduino/blob/intellisense-autoconfig/BRANCHNOTES.md) or poll the readme in the [alpha release Dropbox folder](https://www.dropbox.com/sh/whmcdt26chyjgby/AAB1Ld2fzZ9Z_NfM3CRay17wa).

elektronikworkshop, 2020-02-27:
Ok, I added support for `postbuild` to `0.2.30-alpha13`. Configuration as usual:
```json
{
    "board": "esp32:esp32:esp32doit-devkit-v1",
    "sketch": "test.cpp",
    "port": "/dev/ttyUSB0",
    "configuration": "FlashFreq=80,UploadSpeed=921600,DebugLevel=none",
    "prebuild": "echo \"pre build\"",
    "postbuild": "echo \"post build\""
}
```
Same rules as for `prebuild` apply.
My test (I tested on Linux only up to now):
```
[Starting] Verifying sketch 'test.cpp'
Running pre-build command: "echo "pre build""
pre build
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
Running post-build command: "echo "post build""
post build
IntelliSense configuration already up to date. To manually rebuild your IntelliSense configuration run "Ctrl + Alt + I"
[Done] Verifying sketch 'test.cpp'
```

Regards,
EW

elektronikworkshop, 2020-02-27:
The post build functionality closes this and similar issues which request features some kind of functionality to be performed after and before builds. It will be part of my upcoming pull request.

elektronikworkshop, 2020-02-27:
For the people asking for the `bin` copy: the post build command could be setup like
```JSON
{
    "board": "esp32:esp32:esp32doit-devkit-v1",
    "sketch": "test.cpp",
    "output": "../build",
    "port": "/dev/ttyUSB0",
    "configuration": "FlashFreq=80,UploadSpeed=921600,DebugLevel=none",
    "postbuild": "cp ../build/whatever.bin ./"
}
```
on UNIX systems (Linux, OSX). On Windows you have to use `copy`.

MickGyver, 2020-02-28:
> The post build functionality closes this and similar issues which request features some kind of functionality to be performed after and before builds. It will be part of my upcoming pull request.

Thank you very much! :)

Pantastisch, 2020-02-28:
> For the people asking for the `bin` copy: the post build command could be setup like
> 
> ```json
> {
>     "board": "esp32:esp32:esp32doit-devkit-v1",
>     "sketch": "test.cpp",
>     "output": "../build",
>     "port": "/dev/ttyUSB0",
>     "configuration": "FlashFreq=80,UploadSpeed=921600,DebugLevel=none",
>     "postbuild": "cp ../build/whatever.bin ./"
> }
> ```
> 
> on UNIX systems (Linux, OSX). On Windows you have to use `copy` instead of `cp`.

Great Job!

elektronikworkshop, 2020-02-28:
Hi @Pantastisch @MickGyver 

Thanks for your kudos!

I finished my work on the pre- and post-build command support. They now feature
* Full `cmd` (Windows) and `bash` syntax (Linux and OSX)
* A nice set of environment variables during their invocation

Detailed documentation can be found [here](https://github.com/elektronikworkshop/vscode-arduino/blob/intellisense-autoconfig/README.md#pre--and-post-build-commands).

This allows to run the commands/scripts only during certain builds (e.g. verify, upload) and enables everyone to write pretty powerful pre-/post-build operations (*note to the not so geeky*: the environment variables are also available within scripts you call within the commands, just google `python environment variables` for instance).

Here are some (not so powerful :smile:) examples. The follwing `arduino.json`-setup for Linux/OSX systems
```json
{
    "board": "arduino:avr:nano",
    "sketch": "test.cpp",
    "port": "/dev/ttyUSB0",
    "configuration": "cpu=atmega328old",
    "prebuild": "if [ ${VSCA_BUILD_MODE} == \"Verifying\" ]; then printf \"pre build variables:\\n\\tVSCA_BUILD_MODE=${VSCA_BUILD_MODE}\\n\\tVSCA_SKETCH=${VSCA_SKETCH}\\n\\tVSCA_BOARD=${VSCA_BOARD}\\n\\tVSCA_WORKSPACE_DIR=${VSCA_WORKSPACE_DIR}\\n\\tVSCA_LOG_LEVEL=${VSCA_LOG_LEVEL}\\n\\tVSCA_SERIAL=${VSCA_SERIAL}\\n\\tVSCA_BUILD_DIR=${VSCA_BUILD_DIR}\\n\"; fi",
}
```
will produce during verify
```
pre build variables:
	VSCA_BUILD_MODE=Verifying
	VSCA_SKETCH=test.cpp
	VSCA_BOARD=arduino:avr:nano:cpu=atmega328old
	VSCA_WORKSPACE_DIR=/home/uli/Projects/delme
	VSCA_LOG_LEVEL=info
	VSCA_SERIAL=/dev/ttyUSB0
	VSCA_BUILD_DIR=
```
and remain silent on other builds.

Somthing similar for Windows
```json
{
    "board": "arduino:avr:nano",
    "sketch": "test.ino",
    "configuration": "cpu=atmega328",
    "prebuild": "IF \"%VSCA_BUILD_MODE%\"==\"Verifying\" (echo VSCA_BUILD_MODE=%VSCA_BUILD_MODE% && echo VSCA_BOARD=%VSCA_BOARD%)"
}
```
produces
```
[Starting] Verifying sketch 'test.ino'
Running pre-build command: "IF "%VSCA_BUILD_MODE%"=="Verifying" (echo VSCA_BUILD_MODE=%VSCA_BUILD_MODE% && echo VSCA_BOARD=%VSCA_BOARD%)"
VSCA_BUILD_MODE=Verifying 
VSCA_BOARD=arduino:avr:nano:cpu=atmega328
Loading configuration...
<...>
```
These features are part of the `alpha15` release - to be found in the Dropbox folder above.

Regards,  
EW

Permpol, 2020-05-16:
Yes, I need to use it.