microsoft/vscode-arduino #791: Suddenly can't find arduino IDE

Proladon, 2019-03-18:
Since yesterday, the when i wanna upload the code to board or do others things, it will show 
`Cannot find Arduino IDE. Please specify the "arduino.path" in the User Settings. `

But i have already setting and worked well previously. 
I checked everything settings, it's nothing wrong and i didn't change anything...
like user setting 、 worksapce setting、Folder setting 、Extentsions setting...
every "arduino.path": "D:\\Arduino\\arduino.exe" had set

maybe u will say the arduino path is not in "C:\"
but it did works well previously
is it could be a bug?

Sneezry, 2019-03-22:
Hi @Proladon have you tried `"arduino.path": "D:\Arduino"`?

Proladon, 2019-03-25:
> Hi @Proladon have you tried `"arduino.path": "D:\Arduino"`?

Yeah....didn't works too...

Sneezry, 2019-03-25:
Is there a file named `arduino_debug.exe` in `D:\Arduino` folder?

Proladon, 2019-03-25:
> Is there a file named `arduino_debug.exe` in `D:\Arduino` folder?

yeah, you can see in the image_2. under the arduino.exe

Sneezry, 2019-03-25:
Download the zip file attached, unzip it. Open VS Code, then extension tab, click the three dots on top right cornet, select `Install from VSIX` to install the extension, then run `Arduino Path Dump` command. What result can you get?

[image]

[vscode-arduino-path-missing-validator-0.0.1.vsix.zip](https://github.com/Microsoft/vscode-arduino/files/3002071/vscode-arduino-path-missing-validator-0.0.1.vsix.zip)

Proladon, 2019-03-25:
> Download the zip file attached, unzip it. Open VS Code, then extension tab, click the three dots on top right corner, select `Install from VSIX` to install the extension, then run `Arduino Path Dump` command. What result can you get?

[image]
[image]
[image]

and then it open the Arduino IDE 
[image]

Sneezry, 2019-03-25:
Press `F1` to open command palette, the execute `Arduino Path Dump`.

[image]

Proladon, 2019-03-25:
oh..sorry i thought is terminal command lol

i got this:
```
Path from config: D:\Arduino\arduino.exe
  Directory exists: false
  Arduino CLI exists: false

Path from Reg: D:\Arduino
  Directory exists: true
  Arduino CLI exists: true

Path from where command: D:\Arduino\arduino.exe
  Directory exists: false
  Arduino CLI exists: false

Final path: D:\Arduino\arduino.exe
  Directroy exists: false
  Arduino CLI exists: false

Oops, Arduino extension cannot find Arduino CLI :\
```
so i need to use D:\Arduino ?
but i had already tired this  hm...

Sneezry, 2019-03-25:
You can just remove `arduino.path` from your settings, the extension can automatically find it from the registry.

Proladon, 2019-03-25:
> You can just remove `arduino.path` from your settings, the extension can automatically find it from the registry.

OMG.....it works....really dumb (°ཀ°)
it confused me for two weeks....

Anyway, Thanks you very much dude !!! Your really nice person ~

Sneezry, 2019-03-25:
Great! Then I will close this issue. Please feel free to open a new one if you have any other question.

closed 2019-03-25 by Sneezry

bec3x, 2020-01-16:
Hello! I am having the same issue. I have followed the steps in this thread but it is still not working. Here are my results from the Arduino Path Dump 

[image]

me6ran, 2020-03-22:
Got the same error as @bec3x 
[image]

vukyvt2002, 2020-04-25:
> Got the same error as @bec3x
> [image]

you have to use the folder contained arduino.exe

CarlosA1804, 2020-08-20:
[image]
i have this error

cdebel2005, 2020-09-20:
Just want to precise to some users that if you download the Arduino IDE for **Windows 10**, you will probably end up with a Java version downloaded from the Microsoft Store, which doesn't provide any "arduino.exe".  You got to download the version for Windows 7 of the IDE.  I don't know if there's some way to use the Java version with this extension, i've not searched for that yet.

Technopy311, 2021-06-03:
Hello people, i got a possible solution that worked for me. Like now in windows 10 you must install th IDE from the Microsoft Store, vscode had issues detecting arduino and all that stuff. What i've made to solve the problem, was to download the arduino ide from the official site, but selecting the zip file option, for windows. 

Go to the [Official PAge](https://www.arduino.cc/en/software) and here, choose the "Windows Zip File"  option. I´ve downloaded it, and then i moved the zip file into my documents folder, in it, i extracted the zip file.

Then i went to the vscode settings for the Arduino extension, and in the "Arduino: Path" i entered the path, like this "C:\Users\YOUR-USER\Documents\arduino-1.8.15\" (without the two " )  this can be done with any directory, but keep in mind that the path is without the "arduino.exe", it is just the location. 

Then in the Arduino: Command Path, you just need to put "arduino_debug.exe", but with no path, because the extension uses the path we gave it before. Lastly, you just close vscode, and open it again, it should give no errors.

Magnum164, 2021-07-10:
Wanted to leave a comment I had the same issue, this thread helped but I had to gleam through until I found the problem. My problem was not realizing that the setting arduino.commandPath was actually looking for the single exe for the debug, arduino_debug.exe. I think if MS updated the help info to make that clearer it would resolve this issue for everyone. To me a command path is not the same as a executable file.

Flipped199, 2022-03-24:
Hello, I have the same problem, but after referring to your message, I got the following result, my ide version is 2.0.0.rc5, cli is 0.21.0, is it caused by version incompatibility?
[image]
[image]