microsoft/vscode-arduino #792: Serial Monitor has no Autoscroll option and it stops auto-scrolling once you click on the window

vishnumaiea, 2019-03-20:
Issue Type: <b>Bug</b>

**Problem 1** : There is no explicit option to enable or disable Autoscroll in the available Serial Monitor window.

**Problem 2** : The Serial Monitor window does scroll when you first open it but stops scrolling once you click anywhere inside the window. You need to manually scroll from that point which is very annoying.

**Solution** : Need an explicit option to enable or disable Autoscroll in Serial Monitor window and it must not be affected by when or where the user clicks.

Extension version: 0.2.25
VS Code version: Code 1.32.3 (a3db5be9b5c6ba46bb7555ec5d60178ecc2eaae4, 2019-03-14T23:43:35.476Z)
OS version: Windows_NT x64 6.1.7601

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i3-3110M CPU @ 2.40GHz (4 x 2394)|
|GPU Status|2d_canvas: enabled<br>checker_imaging: disabled_off<br>flash_3d: enabled<br>flash_stage3d: enabled<br>flash_stage3d_baseline: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>native_gpu_memory_buffers: disabled_software<br>rasterization: enabled<br>surface_synchronization: enabled_on<br>video_decode: enabled<br>webgl: enabled<br>webgl2: enabled|
|Memory (System)|7.88GB (1.45GB free)|
|Process Argv||
|Screen Reader|no|
|VM|0%|

</details>

Sneezry, 2019-03-20:
I believe this is a feature of vscode to make it possible to copy selected content in output window.

vishnumaiea, 2019-03-20:
@Sneezry Even if so, having an option to turn on and off Autoscroll wouldn't be a problem right ?

Sneezry, 2019-03-20:
There is. The lock icon above output window is the switch.

vishnumaiea, 2019-03-21:
Oh! My bad! LOL

closed 2019-03-21 by vishnumaiea

cocohub, 2020-03-30:
Even though I click the lock icon, it stops auto scrolling for some reason. Anyone got this error?

Yorams, 2020-05-10:
I've got the same error. Very annoying.

Luis-Agamez, 2022-10-06:
The solution is to click on the two arrows pointing down on the sereal monitor