microsoft/vscode-arduino #796: Extension causes high cpu load

zaid-kamil, 2019-03-27:
- Issue Type: `Performance`
- Extension Name: `vscode-arduino`
- Extension Version: `0.2.25`
- OS Version: `Linux x64 4.15.0-29deepin-generic`
- VSCode version: `1.32.3`

:warning: Make sure to **attach** this file from your *home*-directory: `/home/stormlightx/vsciot-vscode.vscode-arduino-unresponsive.cpuprofile.txt` :warning:

Find more details here: https://github.com/Microsoft/vscode/wiki/Explain:-extension-causes-high-cpu-load

jtsak, 2019-03-30:
I have experienced the same kind of issue.

- Issue Type: `Performance`
- Extension Name: `vscode-arduino`
- Extension Version: `0.2.25`
- OS Version: `Windows_NT x64 10.0.17134`
- VSCode version: `1.32.3`

[vsciot-vscode.vscode-arduino-unresponsive.cpuprofile.txt](https://github.com/Microsoft/vscode-arduino/files/3025137/vsciot-vscode.vscode-arduino-unresponsive.cpuprofile.txt)

czgtest, 2019-04-12:
duplicate with #785 , so close it

closed 2019-04-12 by czgtest

zaid-kamil, 2019-04-12:
Ok