microsoft/vscode-arduino #800: Arduino Board Manage, exit with code 255

nehaunilever, 2019-04-16:
Hi All,

I am new to programming and am trying to install az3166 board. As soon as I open Arduino board manager , I see output screen exit with code 255.
Below is the error in the logs. 
C:\Users\Admin\AppData\Local\Arduino15\package_index.json.tmp: The process cannot access the file because it is being used by another process.

I am trying to navigate to the path mentioned and I cannot find theAppData folder in this location in my C drive.

Can you please help me understand where is this Location picked up from? Also no idea where this package is installed in my pc, how to check this?

cosminpolifronie, 2019-05-04:
I have the same issue on Arch Linux. I have the following error:

> [Starting] Update package index files...
> Picked up JAVA_TOOL_OPTIONS: 
> Loading configuration...
> Initializing packages...
> Preparing boards...
> java.lang.NullPointerException
> 	at processing.app.BaseNoGui.onBoardOrPortChange(BaseNoGui.java:679)
> 	at processing.app.Base.onBoardOrPortChange(Base.java:1317)
> 	at processing.app.Base.<init>(Base.java:284)
> 	at processing.app.Base.main(Base.java:151)
> [Error] Exit with code=255
> 
> [Starting] Update library index files...
> Picked up JAVA_TOOL_OPTIONS: 
> Loading configuration...
> Initializing packages...
> Preparing boards...
> java.lang.NullPointerException
> 	at processing.app.BaseNoGui.onBoardOrPortChange(BaseNoGui.java:679)
> 	at processing.app.Base.onBoardOrPortChange(Base.java:1317)
> 	at processing.app.Base.<init>(Base.java:284)
> 	at processing.app.Base.main(Base.java:151)
> [Error] Exit with code=255

dooriya, 2019-05-05:
Please refer to the installation section in https://microsoft.github.io/azure-iot-developer-kit/docs/faq/

bularond, 2019-05-24:
I installed arduino-1.8.6 and it start working

closed 2019-06-28 by stale[bot]

Wulfheart, 2019-11-18:
> 
> 
> Please refer to the installation section in https://microsoft.github.io/azure-iot-developer-kit/docs/faq/

I'm unsure how this is related to the issue. I'm getting the same error.