microsoft/vscode-arduino #802: Excluding .vscode from sidebar interferes with arduino.json

kedestin, 2019-04-16:
* Version of VS Code and this extension
   VS Code: 1.33.1
    Arduino: Version 0.2.25
 * OS/Platform you are working on
   Windows
 *  Reproducible steps (1... 2... 3...) that cause the issue
Add
```
"**/.vscode": false
````
to list of exclusions in "files.exclude".

Perform any action that would modify arduino.json (select board, port, programmer, etc..)

 * What you expected to see, versus what you actually saw

   * Expected
      
     That selecting board, port and programmer would update the arduino.json file  accordingly
   
    * Saw
      
       Arduino.json was successive overwritten by each action. For example, selecting board type would overwrite everything in arduino.json and fill in data for the board type.  The status bar **would** show selected data as normal, **except** for board type which would stay blank. This seems to prevent uploads to the arduino.

Note: This started immediately after a recent VS Code update

gecko9, 2019-05-01:
I have experienced the same issue but in addition on opening the project none of the settings in arduino.json are loaded into the project. VSCode writes to the the file but doesn't read it.

Sneezry, 2019-05-14:
This is as designed. Arduino extension finds `arduino.json` with VS Code workspace API [`findFiles`](https://code.visualstudio.com/api/references/vscode-api#2292) method. If you exclude `.vscode` folder, Arduino extension couldn't find `arduino.json`, and would create a new one.