microsoft/vscode-arduino #805: [Error] Exit with code=undefined

human2010, 2019-04-26:
Issue Type: <b>Feature Request</b>

[Error] Exit with code=undefined如何解决？

Extension version: 0.2.25
VS Code version: Code 1.33.1 (51b0b28134d51361cf996d2f0a1c698247aeabd8, 2019-04-11T08:27:14.102Z)
OS version: Windows_NT x64 6.1.7601

Sneezry, 2019-04-26:
What's the version of Arduino IDE?

human2010, 2019-04-26:
arduino 1.8.8

human2010, 2019-04-26:
What is the problem? Please also give pointers!

human2010, 2019-04-26:
@Sneezry

Sneezry, 2019-04-26:
What's the result you get with this guide https://github.com/Microsoft/vscode-arduino/issues/791#issuecomment-476089760

human2010, 2019-04-26:
Path from config: <NOT FOUND>
  Directory exists: 
  Arduino CLI exists: 

Path from Reg: C:\Program Files (x86)\Arduino
  Directory exists: true
  Arduino CLI exists: true
The result of running the Arduino Path Dump command：
Path from where command: <NOT FOUND>
  Directory exists: undefined
  Arduino CLI exists: undefined

Final path: C:\Program Files (x86)\Arduino
  Directroy exists: true
  Arduino CLI exists: true

It seems Arduino extension can find Arduino CLI successfully!

Sneezry, 2019-04-26:
Could you provide steps to reproduce this error message?

human2010, 2019-04-26:
[image]
As long as you compile, you will get an error.

human2010, 2019-04-26:
It can be used 1 month ago. I don’t know what happened recently.

human2010, 2019-04-26:
Visual Studio Code and Arduino are reinstalled, and it is not acceptable to change Arduino to 1.8.6.

Sneezry, 2019-04-26:
Does this error only show with Arduino 1.8.8? Does Arduino 1.8.6 work?

human2010, 2019-04-26:
Neither will work

Sneezry, 2019-04-26:
Can you see any other information in dev tool console? You can open dev tool from **Help -> Toggle Developer Tools**.

human2010, 2019-04-26:
[image]

Sneezry, 2019-04-26:
I cannot find any useful information about this issue. Here are some options I think you can have a try.

1. Open Arduino IDE to update board packages and compile the project.

2. Open Arduino Board Manager in VS Code Arduino extension to see if the page load correctly.

3. Arduino CLI may be blocked by Windows firewall. A file called arduino_debug.exe should be in your Arduino IDE installation folder, and VS Code extension calls this file to execute Arduino commands. This file needs to access to the Internet.

closed 2019-05-31 by stale[bot]

aster, 2019-07-02:
I had a similar situation, but I solved it.
I'm not English-speaking, so I'm sorry if my grammar is wrong.

In my case, the setting for the arduino IDE was the cause.
arduinoIDE's
file-> preferences-> settings-> editor language
My initial value: Japanese
New value: English
After changing to this, when I restarted vs code, the log for vs code was displayed correctly.

I think that it was reflected by changing the setting file of arduinoIDE side once.
Please try it.

thangiot, 2019-07-21:
I've got the same problem but I cannot solve my problems with these instructions above. I used Arduino 1.8.9, vscode 1.36.1, Windows 10, Arduino Path: "C:\Program Files (x86)\Arduino" . Thanks
```
[Starting] Verify sketch - Blink.ino
Please see the build logs in Output path: c:\Users\tqtha\OneDrive\Documents\Workspace\Arduino_Projects\build
[Error] Exit with code=undefined
```

arduino.json
```
{
    "sketch": "Blink.ino",
    "port": "COM3",
    "board": "arduino:avr:nano",
    "output": "./build"
}
```

c_cpp_properties.json
```
{
    "configurations": [
        {
            "name": "Win32",
            "includePath": [
                "${workspaceFolder}\\**",
                "C:/Program Files (x86)\\Arduino\\hardware\\tools\\avr\\**",
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\**",
                "C:\\Program Files (x86)\\Arduino\\tools\\**"
            ],
            "forcedInclude": [
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino\\Arduino.h"
            ],
            "intelliSenseMode": "msvc-x64"
        }
    ],
    "version": 4
}
```
settings.json
```
{
    "arduino.path": "C:\\Program Files (x86)\\Arduino", 
    "arduino.additionalUrls": "", 
    "arduino.logLevel": "info",
    "arduino.commandPath": "arduino_debug.exe"
}
```