microsoft/vscode-arduino #807: change upload sketch

5shekel, 2019-04-27:
hoepfully its my lack of reading.

i am trying to change arduino.sketch to a different one in the workspace.
aside from manually changing 
  ``` "sketch": "smoothed-calib\\calib.ino",```

i cant find a  way.

lacerto, 2019-06-01:
I also use workspaces with multiple subdirectories containing sketches and did not find a way to change the main sketch either.

If you delete "sketch" in arduino.json and try to verify or upload a sketch then the extension lets you select it from a list.
So the feature is there, it just needs to be added to the command palette, I think.

As a workaround I just wrote a task to set the value of "sketch" to the currently open file in the editor.

ShixiangWang, 2019-06-13:
I have the same issue. I cannot change sketch when changing to another .ino file.

danielsiwiec, 2019-10-05:
Same - this would be a nice feature to have. Perhaps an action like `Upload current file`

MatthewSteeves, 2020-01-06:
> I also use workspaces with multiple subdirectories containing sketches and did not find a way to change the main sketch either.
>
> If you delete "sketch" in arduino.json and try to verify or upload a [...]

Good idea re: workaround using task! Care to share the details?

lacerto, 2020-01-06:
> Good idea re: workaround using task! Care to share the details?

Sure. It's nothing fancy just a Python script overwriting arduino.json:
https://gist.github.com/lacerto/2a0e7f0102c02e5d01784573abc43736

The same can surely be accomplished with a couple of built-in command line utilities, maybe using [jq](https://stedolan.github.io/jq/) or some other command line json processor. It was easier for me with Python.

jgoeders, 2020-01-24:
> Sure. It's nothing fancy just a Python script overwriting arduino.json:
> https://gist.github.com/lacerto/2a0e7f0102c02e5d01784573abc43736
>
> The same can surely be accomplished with a couple of [...]

Thanks, this worked well.  I had to change it to  "command": "python set_main_sketch.py" to get it to work in Windows.  Also, although it changes, the status bar the bottom of VS Code doesn't reflect the change.  I'm not sure how to trigger a refresh there.

maddogjt, 2020-10-30:
PR #1128 should cover this