microsoft/vscode-arduino #808: identifier "Serial" is undefined

loongEgg, 2019-04-29:
My VS Code have a problem:identifier "Serial" is undefined
but I upload program to my arduino nano OK.
this is my arduino file:
``` c
void setup(){
    pinMode(LED_BUILTIN, OUTPUT);
    Serial.begin(9600);
}

void loop(){
    digitalWrite(LED_BUILTIN, HIGH);
    Serial.println("hello arduino 2333333333333");
    delay(1000);
    digitalWrite(LED_BUILTIN, LOW);
    delay(1000);
}
```

this is arduino.json
``` json
{
    "board": "arduino:avr:nano",
    "configuration": "cpu=atmega328old",
    "programmer": "AVRISP mkII",
    "sketch": "serialPort.ino",
    "port": "COM9"
}
```

this is c_cpp_properties.json
```  json
{
    "configurations": [
        {
            "name": "Win32",
            "includePath": [
                "C:\\\\Arduino\\\\tools\\\\**",
                "C:\\\\Arduino\\\\hardware\\\\arduino\\\\avr\\\\**",
                "C:\\\\Arduino\\\\hardware\\\\arduino\\\\avr\\\\cores\\\\arduino\\\\**",
                "C:\\\\Arduino\\\\hardware\\\\tools\\\\avr\\\\avr\\\\include\\\\avr\\\\**",
                "C:\\\\Arduino\\\\hardware\\\tools\\\\**", 
                "C:\\\\Arduino\\\\hardware\\\tools\\\\avr\\\\avr\\\\include\\\\**"
            ],
            "forcedInclude": [
                "C:\\\\Arduino\\\\hardware\\\\arduino\\\\avr\\\\cores\\\\arduino\\\\Arduino.h"
            ],
            "intelliSenseMode": "gcc-x64",
            "compilerPath": "C:/mingw64/bin/gcc.exe",
            "cStandard": "c11",
            "cppStandard": "c++17"
        }
    ],
    "version": 4
}
```

dilin-MS, 2019-05-05:
Hi, could you please provide more information of your situation? Where does the message `identifier "Serial" is undefined` occur? Is it an intelliSense prompt message(like a red wavy line suggesting error) or an output message when you try to compile the code?
Since you say that you upload the code to arduino nano device and the code can work well, you mean that the code can compile smoothly without error and can work well. But the vscode intelliSense just gives out the undefined message. Did I misunderstand what you mean?

alibas-eee, 2019-05-05:
i have same error like this.
i try to include hardwareserial but give errors when writing include files too
[image]

[image]

gemul, 2019-05-07:
I also have the same error.
[image]
commands like delay, pinMode, digitalWrite, digitalRead, etc. works fine with intellisense.
but when using Serial, it shows: 
`identifier "Serial" is undefined`

trying to add `#include<HardwareSerial.h>` it gives me:
`#include errors detected. Please update your includePath. Squiggles are disabled for this translation unit (F:\project\arduino\sketches\gto\app.ino).`
[image]

and then after clicking on the lightbulb, then "add to includePath..." i get :
`PCH warning: header stop cannot be in a linkage block.  An IntelliSense PCH file was not generated.`
[image]

and this is my c_cpp_properties.json:
```
{
    "configurations": [
        {
            "name": "Win32",
            "includePath": [
                "C:/Users/gemul/AppData/Local/Arduino15/packages/arduino/tools/**",
                "C:/Users/gemul/AppData/Local/Arduino15/packages/arduino/hardware/avr/1.6.23/**",
                "C:/Users/gemul/AppData/Local/Arduino15/packages/arduino/hardware/avr/1.6.23/cores/arduino"
            ],
            "forcedInclude": [
                "C:/Users/gemul/AppData/Local/Arduino15/packages/arduino/hardware/avr/1.6.23/cores/arduino/Arduino.h"
            ],
            "intelliSenseMode": "msvc-x64",
            "compilerPath": "C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx64/x64/cl.exe",
            "cStandard": "c11",
            "cppStandard": "c++17"
        }
    ],
    "version": 4
}
```
the line:
`C:/Users/gemul/AppData/Local/Arduino15/packages/arduino/hardware/avr/1.6.23/cores/arduino`
is added when i clicked on lightbulb then "add to includePath...".

By the way, using SoftwareSerial works fine.
[image]
it added
`"C:/Users/gemul/AppData/Local/Arduino15/packages/arduino/hardware/avr/1.6.23/libraries/SoftwareSerial/src"`
in includePath.

gemul, 2019-05-08:
It seems i have solved my own issue. And here is what i have tried.

First, i can get rid of the intellisense errors by simply using tag parser instead of default for my cpp intellisense engine. by changing **C_Cpp:Intelli Sense Engine** in **File>preference>settings** to "**Tag Parser**".
That indeed get rid of the errors, and put intellisense autocompletion and suggestion to works again. But it get rid of error checking (which is very important for me).
So if you don't mind using tag parser, then go on.

Second, after googling for a while, i found out that in order for **Serial** to be able to works with intellisense, you need to add `"defines": [ "USBCON" ]` in your **c_cpp_properties.json**. And i did, and now the intellisense is working (**i changed back the cpp intellisense engine to Default**).
So you might try this one first, perhaps it will solve your problem. If not, use the tag parser one.
my c_cpp_properties.json. now looks like this:
[image]

WilsonCazarre, 2019-05-23:
> It seems i have solved my own issue. And here is what i have tried.
>
> First, i can get rid of the intellisense errors by simply using tag parser instead of default for my cpp intellisense engine. by [...]

Thanks for this answer man, now my IntelliSense is working very well, code completion and the libraries :)

HollisAnthony, 2019-05-24:
> It seems i have solved my own issue. And here is what i have tried.
>
> First, i can get rid of the intellisense errors by simply using tag parser instead of default for my cpp intellisense engine. by [...]

Extremely appreciative. Thank you.

gregorthebigmac, 2019-06-06:
@gemul Dude, THANK YOU!!!!!! I spent entirely too long tracking this down, and finally someone solved it! I just confirmed this problem exists on Linux, as well, and your solution fixed it! Finally got my Linter back!!!!

maludwig, 2019-06-12:
@dilin-MS this bug still persists in my version of the extension. The workaround described above fixes it, but should we Close this Issue before the bug itself is fixed?

gregorthebigmac, 2019-06-14:
> @dilin-MS this bug still persists in my version of the extension. The workaround described above fixes it, but should we Close this Issue before the bug itself is fixed?

Agreed. If it were my software, I wouldn't be closing the bug report unless we'd already fixed it, and were prepping to roll it out in the next update, and even then, I'd probably wait a few days until after the update gets pushed out to make sure we don't get any more complaints about it before calling it closed.

deandreamatias, 2019-06-26:
Hi people.
I have a similar problem with Serial2 and Serial3. They remain undefined. Any suggestions to fix this?

fpistm, 2019-06-26:
I got the same issue.
My comprehension is that the vscode arduino extension is not able to get definition defined in the `platform.txt`/`boards.txt`. So they are not added to C/C++ properties automatically. So, they have to be manually added to the `.vscode/c_cpp_properties.json`

Example for this core for an STM32F1 based board (BluePillF103C8): 
https://github.com/stm32duino/Arduino_Core_STM32

```json
            "defines": [
                "ARDUINO=10809",
                "HAL_UART_MODULE_ENABLED",
                "STM32F1xx"
            ]
```
`HAL_UART_MODULE_ENABLED` is manged thanks the Arduino menu and require to get Serial instance defined for C/C++ code browsing.
So, you have to find which define is required.

In the same way, the generated `includePath` in the `.vscode/c_cpp_properties.json` is generic and include all top level folders:
```
            "includePath": [
                "${workspaceRoot}",
                "c:\\\\STM32\\\\arduino\\\\arduino-1.8.9-windows\\\\arduino-1.8.9\\\\portable\\\\packages\\\\STM32\\\\hardware\\\\stm32\\\\1.6.0\\\\**",
                "c:\\\\STM32\\\\arduino\\\\arduino-1.8.9-windows\\\\arduino-1.8.9\\\\portable\\\\packages\\\\STM32\\\\tools\\\\**",
                "c:\\\\STM32\\\\arduino\\\\arduino-1.8.9-windows\\\\arduino-1.8.9\\\\hardware\\\\tools\\\\**",
                "c:\\\\STM32\\\\arduino\\\\arduino-1.8.9-windows\\\\arduino-1.8.9\\\\libraries\\\\**",
                "C:\\\\Users\\\\frq92243\\\\Documents\\\\Arduino\\\\hardware\\\\tools\\\\**",
                "C:\\\\Users\\\\frq92243\\\\Documents\\\\Arduino\\\\libraries\\\\**"
            ],
```

But all headers should not be parsed and could depend of definitions/options, 
In my case, some path are added to include path thanks the `platform.txt` 

https://github.com/stm32duino/Arduino_Core_STM32/blob/1bd6ffaf8a594b7224c3789642aad9c29a174271/platform.txt#L12

`compiler.stm.extra_include="-I{build.source.path}" "-I{build.core.path}/avr" "-I{build.core.path}/stm32" "-I{build.core.path}/stm32/LL" "-I{build.core.path}/stm32/usb" "-I{build.core.path}/stm32/usb/hid" "-I{build.core.path}/stm32/usb/cdc" "-I{build.system.path}/Drivers/{build.series}_HAL_Driver/Inc/" "-I{build.system.path}/Drivers/{build.series}_HAL_Driver/Src/" "-I{build.system.path}/{build.series}/" "-I{build.system.path}/Middlewares/ST/STM32_USB_Device_Library/Core/Inc" "-I{build.system.path}/Middlewares/ST/STM32_USB_Device_Library/Core/Src"
`
Here, `{build.series}` is defined thanks the `boards.txt` depending of the board selected.
So, I have to update `includePath` property to get only the right include and avoid multiple definitions.
```
            "includePath": [
                "${workspaceRoot}",
                "c:\\\\STM32\\\\arduino\\\\arduino-1.8.9-windows\\\\arduino-1.8.9\\\\portable\\\\packages\\\\STM32\\\\hardware\\\\stm32\\\\1.6.0\\\\cores\\\\**",
                "c:\\\\STM32\\\\arduino\\\\arduino-1.8.9-windows\\\\arduino-1.8.9\\\\portable\\\\packages\\\\STM32\\\\hardware\\\\stm32\\\\1.6.0\\\\system\\\\STM32F1xx\\\\**",
                "c:\\\\STM32\\\\arduino\\\\arduino-1.8.9-windows\\\\arduino-1.8.9\\\\portable\\\\packages\\\\STM32\\\\hardware\\\\stm32\\\\1.6.0\\\\system\\\\Drivers\\\\STM32F1xx_HAL_Driver\\\\**",
                "c:\\\\STM32\\\\arduino\\\\arduino-1.8.9-windows\\\\arduino-1.8.9\\\\portable\\\\packages\\\\STM32\\\\hardware\\\\stm32\\\\1.6.0\\\\system\\\\Drivers\\\\CMSIS\\\\Device\\\\ST\\\\STM32F1xx\\\\**",
                "c:\\\\STM32\\\\arduino\\\\arduino-1.8.9-windows\\\\arduino-1.8.9\\\\portable\\\\packages\\\\STM32\\\\hardware\\\\stm32\\\\1.6.0\\\\variants\\\\BLUEPILL_F103XX\\\\**",
                "c:\\\\STM32\\\\arduino\\\\arduino-1.8.9-windows\\\\arduino-1.8.9\\\\portable\\\\packages\\\\STM32\\\\tools\\\\**",
                "c:\\\\STM32\\\\arduino\\\\arduino-1.8.9-windows\\\\arduino-1.8.9\\\\hardware\\\\tools\\\\**",
                "c:\\\\STM32\\\\arduino\\\\arduino-1.8.9-windows\\\\arduino-1.8.9\\\\libraries\\\\**"
            ],
```

With those updates, there is no more error reported and code browsing is well performed else I've multiple definition.

Note that those errors only rely on code browsing, Arduino build is well performed with correct paths and definitions.

Hope this help and clear.

iFreilicht, 2019-07-08:
It seems having this defined should also be part of #438

55thSwiss, 2019-08-30:
@gemul nice find, this was driving me crazy

drummsb, 2019-09-28:
> It seems i have solved my own issue. And here is what i have tried.
>
> First, i can get rid of the intellisense errors by simply using tag parser instead of default for my cpp intellisense engine. by [...]

This is an amazing find and I now know how to use the define field in the c_cpp_properties.json file. Worked like a champ!  I am greatful!

sakshamsneh, 2019-10-02:
> It seems i have solved my own issue. And here is what i have tried.
>
> First, i can get rid of the intellisense errors by simply using tag parser instead of default for my cpp intellisense engine. by [...]

This is very helpful. I started Arduino coding in VSCode, and encountered this issue almost immediately. Thank you very much.

Sallaa, 2019-11-06:
> It seems i have solved my own issue. And here is what i have tried.
>
> First, i can get rid of the intellisense errors by simply using tag parser instead of default for my cpp intellisense engine. by [...]

Had a similar issue and this solved my problem too. Thank you @gemul.

elektronikworkshop, 2020-02-05:
Hi to everyone who is suffering from this!

I was so annoyed, that I began to work on an auto-configuration mechanism, which parses the build output from Arduino and creates a `c_cpp_properties.json` from it. The goal is that no one wants likely ever edit this file manually again - except for very weird setups for which I will add a configuration flag/option that will keep the extension from touching the file.

I coded some proof of concept already but I haven't committed anything yet. But I began documenting the project [here](https://github.com/elektronikworkshop/vscode-arduino/blob/intellisense-autoconfig/BRANCHNOTES.md).

Anyone who wants to keep me motivated can chip in some :beer:-money - the donation button can be found in the link above.

Other support is appreciated and comprises
* people with Windows and OSX setups to test the cross platform compatibility
* the original maintainers could help me with CI and how to build test-releases

The progress can be followed within the same file. The [repo](https://github.com/elektronikworkshop/vscode-arduino) is set to this feature branch by default.

nachogarmilla, 2020-03-22:
Thank you for sharing. I had same issue.

helviorubens, 2020-04-28:
I had the same issue, so I tried the @gemul 's solution but didn't work for me. However, I've added `#include <Arduino.h>` and worked well.

kopiro, 2020-04-30:
Should just `#include<Arduino.h>` be the solution?

helviorubens, 2020-05-14:
@kopiro I keep doing this, so I restart the VS Code and it works without error messages.

dhiyaaulauliyaa, 2020-05-16:
> It seems i have solved my own issue. And here is what i have tried.
>
> First, i can get rid of the intellisense errors by simply using tag parser instead of default for my cpp intellisense engine. by [...]

this have solved my problem. thank you!

mich2k, 2020-06-01:
Same here, thanks

mich2k, 2020-06-04:
Emh, adding the port parameter is not a fix, but a workaround, I managed too to resolve this with the parameter 
But again, is not a fix

thinkOfaNumber, 2020-08-27:
I found I needed more than just USBCON. When you compile your project, look for a line like this:

`"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=8000000L -DARDUINO=10812 -DARDUINO_AVR_PRO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\eightanaloginputs" "path\\to\\your\\sketch\\sketch.ino.cpp" -o nul
`

Look for the arguments that start with -D... Each one of those is a #define that is set by the arduino build process. In my above example they are `-DF_CPU=8000000L -DARDUINO=10812 -DARDUINO_AVR_PRO -DARDUINO_ARCH_AVR`. Take these (remove the -D) and put then into your "defines" section, like this:

            "defines": [
                "USBCON",
                "F_CPU=8000000L",
                "ARDUINO=10812",
                "ARDUINO_AVR_PRO",
                "ARDUINO_ARCH_AVR"
            ]

iFreilicht, 2020-09-21:
After using elektronikworkshop/vscode-arduino fork with the autogeneration feature, I can attest that this does indeed solve this problem perfectly.

stormbytes, 2020-09-26:
> Should just `#include<Arduino.h>` be the solution?

Well, I tried this. It works. 
Can someone explain what is the issue here?

Dnllns, 2020-10-13:
i have the same problem with the cudaDeviceProp sctruct (Cuda toolkit)
anyone knows how to solve it?

adiazulay, 2020-12-11:
This is fixed in v0.3.4

closed 2020-12-11 by adiazulay

GF-Huang, 2021-02-22:
> This is fixed in v0.3.4

Not fixed, v0.3.5 issue still.

[image]

[image]

[image]

iFreilicht, 2021-02-22:
This will be fixed, soon. If you want to test that, download the [v0.4.0-rc2.vsix](https://github.com/microsoft/vscode-arduino/releases/download/v0.4.0-rc2/vscode-arduino-0.4.0-rc2.vsix) file, and install it in vscode by opening the file via "File" -> "Open File..." or running code from the terminal like `code Downloads/vscode-arduino-0.4.0-rc2.vsix`.

After this, the `c_cpp_properties.json` should be generated automatically, and the `ARDUINO` macro should be defined correctly.

Note that putting `defines: [ "ARDUINO" ]` is not guaranteed to always work! It should be `defines: [ "ARDUINO=10812" ]`.

GF-Huang, 2021-02-22:
`"ARDUINO=10812"` will still cause `uint8_t` not defined.

[image]

iFreilicht, 2021-02-22:
@GF-Huang What board are you using?

GF-Huang, 2021-02-22:
RF-NANO, but this does not effect compile and run, just a syntax error.

iFreilicht, 2021-02-28:
This should be fixed in v0.4.0, so please try to download and install that as I described above :)

GF-Huang, 2021-03-09:
v0.4.0-rc3 still not include `ARDUINO` macro.

[image]

`.vscode\c_cpp_properties.json`

```json
{
    "version": 4,
    "configurations": [
        {
            "name": "Win32",
            "compilerPath": "C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29333/bin/Hostx64/x64/cl.exe",
            "compilerArgs": [],
            "intelliSenseMode": "windows-msvc-x64",
            "includePath": [
                "D:\\Code\\Arduino\\libraries\\arduino_690070",
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\SPI\\src",
                "C:\\Program Files (x86)\\Arduino\\tools\\**",
                "D:\\Code\\Arduino\\libraries\\**",
                "C:\\Program Files (x86)\\Arduino\\libraries\\**",
                "C:\\Program Files (x86)\\Arduino\\hardware\\tools\\**",
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\**"
            ],
            "forcedInclude": [
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino\\Arduino.h"
            ],
            "cStandard": "c17",
            "cppStandard": "c++17",
            "defines": [
                "USBCON"
            ]
        },
        {
            "name": "Arduino",
            "compilerPath": "C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr\\bin\\avr-g++",
            "compilerArgs": [
                "-w",
                "-std=gnu++11",
                "-fpermissive",
                "-fno-exceptions",
                "-ffunction-sections",
                "-fdata-sections",
                "-fno-threadsafe-statics",
                "-Wno-error=narrowing"
            ],
            "intelliSenseMode": "gcc-x64",
            "includePath": [
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino",
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\eightanaloginputs",
                "D:\\Code\\Arduino\\libraries\\RF24",
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\SPI\\src",
                "c:\\program files (x86)\\arduino\\hardware\\tools\\avr\\lib\\gcc\\avr\\7.3.0\\include",
                "c:\\program files (x86)\\arduino\\hardware\\tools\\avr\\lib\\gcc\\avr\\7.3.0\\include-fixed",
                "c:\\program files (x86)\\arduino\\hardware\\tools\\avr\\avr\\include"
            ],
            "forcedInclude": [
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino\\Arduino.h"
            ],
            "cStandard": "c11",
            "cppStandard": "c++11",
            "defines": [
                "F_CPU=16000000L",
                "ARDUINO=10813",
                "ARDUINO_AVR_NANO",
                "ARDUINO_ARCH_AVR",
                "__DBL_MIN_EXP__=(-125)",
                "__HQ_FBIT__=15",
                "__cpp_attributes=200809",
                "__UINT_LEAST16_MAX__=0xffffU",
[... 474 lines cut ...]
                "__ULACCUM_EPSILON__=0x1P-32ULK",
                "__cpp_user_defined_literals=200809",
                "__SACCUM_IBIT__=8",
                "__GCC_ATOMIC_LLONG_LOCK_FREE=1",
                "__LDBL_DIG__=6",
                "__FLT_DECIMAL_DIG__=9",
                "__UINT_FAST16_MAX__=0xffffU",
                "__GCC_ATOMIC_SHORT_LOCK_FREE=1",
                "__BUILTIN_AVR_ABSHK=1",
                "__BUILTIN_AVR_FLASH_SEGMENT=1",
                "__INT_LEAST64_WIDTH__=64",
                "__ULLFRACT_MAX__=0XFFFFFFFFFFFFFFFFP-64ULLR",
                "__UINT_FAST8_TYPE__=unsigned char",
                "__USFRACT_EPSILON__=0x1P-8UHR",
                "__ULACCUM_FBIT__=32",
                "__QQ_IBIT__=0",
                "__cpp_init_captures=201304",
                "__ATOMIC_ACQ_REL=4",
                "__ATOMIC_RELEASE=3",
                "__BUILTIN_AVR_FMUL=1"
            ]
        }
    ]
}
```

adiazulay, 2021-03-09:
Sorry the file was misnamed. 
I'm rerunning the build now and the correct file name should display.

belousa, 2021-04-09:
Seems like the auto-configuration does not add the `USBCON` macro. Is there a reason for that?

iFreilicht, 2021-04-10:
> Seems like the auto-configuration does not add the `USBCON` macro. Is there a reason for that?

It should do that for all chips that actually have USB connectivity (Atmega32u4 etc.). What board are you seeing this issue with?

belousa, 2021-04-10:
I have Adafruit Metro. My arduino.json is, however, 
```
{
    "board": "arduino:avr:uno",
    "sketch": "logo.ino",
    "port": "COM3"
}
```

cnbrenci, 2021-04-10:
I'm also seeing the auto-configuration not add `USBCON` on a board with USB

vscode-arduino v0.4.0

arduino.json
```
{
    "configuration": "cpu=8MHzatmega328",
    "board": "arduino:avr:pro",
    "port": "COM3",
    "programmer": "arduino:avrispmkii",
    "sketch": "firmware\\sketches\\FrontTest\\FrontTest.ino"
}
```

c_cpp_properties.json
```
{
    "version": 4,
    "configurations": [
        {
            "name": "Arduino",
            "compilerPath": "C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr\\bin\\avr-g++",
            "compilerArgs": [
                "-std=gnu++11",
                "-fpermissive",
                "-fno-exceptions",
                "-ffunction-sections",
                "-fdata-sections",
                "-fno-threadsafe-statics",
                "-Wno-error=narrowing"
            ],
            "intelliSenseMode": "gcc-x64",
            "includePath": [
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino",
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\eightanaloginputs",
                "C:\\Users\\Cassi\\OneDrive\\Documents\\Arduino\\libraries\\EverydayCalendar",
                "C:\\Users\\Cassi\\OneDrive\\Documents\\Arduino\\libraries\\IQS5xx",
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\SPI\\src",
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\EEPROM\\src",
                "C:\\Users\\Cassi\\OneDrive\\Documents\\Arduino\\libraries\\Wire_BigBuffer",
                "c:\\program files (x86)\\arduino\\hardware\\tools\\avr\\lib\\gcc\\avr\\7.3.0\\include",
                "c:\\program files (x86)\\arduino\\hardware\\tools\\avr\\lib\\gcc\\avr\\7.3.0\\include-fixed",
                "c:\\program files (x86)\\arduino\\hardware\\tools\\avr\\avr\\include"
            ],
            "forcedInclude": [
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino\\Arduino.h"
            ],
            "cStandard": "c11",
            "cppStandard": "c++11",
            "defines": [
                "F_CPU=8000000L",
                "ARDUINO=10813",
                "ARDUINO_AVR_PRO",
                "ARDUINO_ARCH_AVR",
                "__DBL_MIN_EXP__=(-125)",
                "__HQ_FBIT__=15",
                "__cpp_attributes=200809",
                "__UINT_LEAST16_MAX__=0xffffU",
                "__ATOMIC_ACQUIRE=2",
                "__SFRACT_IBIT__=0",
                "__FLT_MIN__=1.17549435e-38F",
                "__GCC_IEC_559_COMPLEX=0",
                "__BUILTIN_AVR_SLEEP=1",
                "__BUILTIN_AVR_COUNTLSULLK=1",
                "__cpp_aggregate_nsdmi=201304",
                "__BUILTIN_AVR_COUNTLSULLR=1",
                "__UFRACT_MAX__=0XFFFFP-16UR",
                "__UINT_LEAST8_TYPE__=unsigned char",
                "__DQ_FBIT__=63",
[... 463 lines cut ...]
                "__ULACCUM_EPSILON__=0x1P-32ULK",
                "__cpp_user_defined_literals=200809",
                "__SACCUM_IBIT__=8",
                "__GCC_ATOMIC_LLONG_LOCK_FREE=1",
                "__LDBL_DIG__=6",
                "__FLT_DECIMAL_DIG__=9",
                "__UINT_FAST16_MAX__=0xffffU",
                "__GCC_ATOMIC_SHORT_LOCK_FREE=1",
                "__BUILTIN_AVR_ABSHK=1",
                "__BUILTIN_AVR_FLASH_SEGMENT=1",
                "__INT_LEAST64_WIDTH__=64",
                "__ULLFRACT_MAX__=0XFFFFFFFFFFFFFFFFP-64ULLR",
                "__UINT_FAST8_TYPE__=unsigned char",
                "__USFRACT_EPSILON__=0x1P-8UHR",
                "__ULACCUM_FBIT__=32",
                "__QQ_IBIT__=0",
                "__cpp_init_captures=201304",
                "__ATOMIC_ACQ_REL=4",
                "__ATOMIC_RELEASE=3",
                "__BUILTIN_AVR_FMUL=1"
            ]
        }
    ]
}
```

iFreilicht, 2021-04-12:
This should be fixed with the next release, see #1216.

Peter707-try, 2021-04-18:
0.4.1.rc1 works fine for me: USBCON is automatically added to c_cpp_properties.json
The Serial (not defined issue is solved.
Thank You very much :) +1:

kakaooo, 2021-06-21:
I'm still having the same issue. I am using LVGL libraries through the Platformio yet the problem is the same. Serial is undefined. Tried to apply both solutions of @gemul ' s solutions. Tried to use the tag parser then the editor recognizes the Serial command but it cannot be compiled within the same issue. However, tried to add "USBCON" command into my c_cpp_properties.json file but no change still receiving the same error message. What am i missing? Any ideas on how to fix this?

I am using a ready platform board from Yeacreate (Nscreen32) it powered by ESP32- WROVER-IE board. The other functions such as digitalWrite, pinMode etc are working just fine. But not for Serial command. 
[image]
[image]
[image]

MarcelloTheArcane, 2021-09-09:
If you want Intellisense autocomplete, instead of `"USBCON"` add `"UBRRH"`:

[image]

[image]

[Source](https://github.com/microsoft/vscode-cpptools/issues/743#issuecomment-302916021)

Revel8804, 2022-05-25:
I am getting this error now. 
[image]
[image]

`            "name": "Arduino",
            "compilerPath": "C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr\\bin\\avr-g++",
            "compilerArgs": [
                "-w",
                "-std=gnu++11",
                "-fpermissive",
                "-fno-exceptions",
                "-ffunction-sections",
                "-fdata-sections",
                "-fno-threadsafe-statics",
                "-Wno-error=narrowing"
            ],
            "intelliSenseMode": "gcc-x64",
            "includePath": [
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino",
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\leonardo",
                "C:\\Users\\Sonic\\Documents\\Arduino\\libraries\\Servo\\src",
                "c:\\program files (x86)\\arduino\\hardware\\tools\\avr\\lib\\gcc\\avr\\7.3.0\\include",
                "c:\\program files (x86)\\arduino\\hardware\\tools\\avr\\lib\\gcc\\avr\\7.3.0\\include-fixed",
                "c:\\program files (x86)\\arduino\\hardware\\tools\\avr\\avr\\include"
            ],
            "forcedInclude": [
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino\\Arduino.h"
            ],
            "cStandard": "c11",
            "cppStandard": "c++11",
            "defines": [
                "F_CPU=16000000L",
                "ARDUINO=10819",
                "ARDUINO_AVR_LEONARDO",
                "ARDUINO_ARCH_AVR",
                "USB_VID=0x2341",
                "USB_PID=0x8036",
                "USB_MANUFACTURER=\"Unknown\"",
                "USB_PRODUCT=\"Arduino Leonardo\"",
                "__DBL_MIN_EXP__=(-125)",
                "__HQ_FBIT__=15",
                "__cpp_attributes=200809",
                "__UINT_LEAST16_MAX__=0xffffU",
                "__ATOMIC_ACQUIRE=2",
                "__SFRACT_IBIT__=0",
                "__FLT_MIN__=1.17549435e-38F",
                "__GCC_IEC_559_COMPLEX=0",
                "__BUILTIN_AVR_SLEEP=1",
                "__BUILTIN_AVR_COUNTLSULLK=1",
                "__cpp_aggregate_nsdmi=201304",
                "__BUILTIN_AVR_COUNTLSULLR=1",
                "__UFRACT_MAX__=0XFFFFP-16UR",
                "__UINT_LEAST8_TYPE__=unsigned char",
                "__DQ_FBIT__=63",
                "__INTMAX_C(c)=c ## LL",
                "__ULFRACT_FBIT__=32",
                "__SACCUM_EPSILON__=0x1P-7HK",
                "__CHAR_BIT__=8",
                "__USQ_IBIT__=0",
                "__UINT8_MAX__=0xff",
                "__ACCUM_FBIT__=15",
                "__WINT_MAX__=0x7fff",
                "__FLT32_MIN_EXP__=(-125)",
                "__cpp_static_assert=200410",
                "__USFRACT_FBIT__=8",
                "__ORDER_LITTLE_ENDIAN__=1234",
                "__SIZE_MAX__=0xffffU",
                "__WCHAR_MAX__=0x7fff",
                "__LACCUM_IBIT__=32",
                "__DBL_DENORM_MIN__=double(1.40129846e-45L)",
[... 445 lines cut ...]
                "__BUILTIN_AVR_COUNTLSUR=1",
                "__SIZEOF_LONG_LONG__=8",
                "__ULACCUM_EPSILON__=0x1P-32ULK",
                "__cpp_user_defined_literals=200809",
                "__SACCUM_IBIT__=8",
                "__GCC_ATOMIC_LLONG_LOCK_FREE=1",
                "__LDBL_DIG__=6",
                "__FLT_DECIMAL_DIG__=9",
                "__UINT_FAST16_MAX__=0xffffU",
                "__GCC_ATOMIC_SHORT_LOCK_FREE=1",
                "__BUILTIN_AVR_ABSHK=1",
                "__BUILTIN_AVR_FLASH_SEGMENT=1",
                "__INT_LEAST64_WIDTH__=64",
                "__ULLFRACT_MAX__=0XFFFFFFFFFFFFFFFFP-64ULLR",
                "__UINT_FAST8_TYPE__=unsigned char",
                "__USFRACT_EPSILON__=0x1P-8UHR",
                "__ULACCUM_FBIT__=32",
                "__QQ_IBIT__=0",
                "__cpp_init_captures=201304",
                "__ATOMIC_ACQ_REL=4",
                "__ATOMIC_RELEASE=3",
                "__BUILTIN_AVR_FMUL=1",
                "USBCON"
            ]
`

krystianzun, 2022-07-21:
Despite having `"USBCON"` automatically added to `c_cpp_properties.json` I'm still having the same issue as some people above. 

Any suggestions on how to tackle this? 

[image]
[image]

chriscummings, 2024-01-21:
Problem persists on m1 mac with vs code-1.85.2 and arduino plugin-0.6.0.

andresmoschini, 2024-03-29:
> Problem persists on m1 mac with vs code-1.85.2 and arduino plugin-0.6.0.

+1