microsoft/vscode-arduino #812: consol throws: "java.lang.NullPointerException thrown while loading gnu.io.RXTXCommDriver" when trying to install Arduino AVR Boards

toalba, 2019-05-08:
Console is throwing an Exception when I am trying to install Arduino AVR Boards from Arduino Board Manager. Arduino is installed via console.

Using:
Ubuntu 18.4
Arduino 1.8.9 (last update) 
Visual Studio Code 1.33

Thrown Exception: java.lang.NullPointerException thrown while loading gnu.io.RXTXCommDriver

Sneezry, 2019-05-09:
Does this link help? https://askubuntu.com/questions/1034173/arduino-serial-communication-uploader-not-working

toalba, 2019-05-09:
No, instructions on this link are not help. I will try to reproduce this issue on a newer Ubuntu edition (19.04)

toalba, 2019-05-12:
How to solve that on a 18.04:
install:
$ sudo apt install oracle-java11-set-default
or
$ sudo apt install oracle-java12-set-default
(when license button stucks try it with shift +enter)

Installing a fresh Arduino Version with(first remove the old on):

$ sudo apt-get install ubuntu-make
$ umake ide arduino
$ sudo usermod -a -G dialout $USER

closed 2019-05-12 by toalba

jpablo27, 2019-06-16:
I had the same problem. 
I installed Java and had same problem again.

What worked for me was uninstalling arduino and then reinstaling it by downloading it from https://www.arduino.cc