microsoft/vscode-arduino #816: Clicking Select Serial port does nothing

C0rn3j, 2019-05-11:
Seems to be slightly related to #664 
Clicking the button does nothing, no menu pops up.

I had to manually add a line
```
"port": "/dev/ttyUSB0"
```
to ./vscode/arduino.json, after which at least upload works, but opening serial monitor fails with:

```
Failed to open serial port /dev/ttyUSB0 due to error: + TypeError: Cannot read property 'close' of null
```

OS: Arch Linux
code 1.33.1
arduino 1.8.9

bmo-at, 2019-05-12:
First of all, disclaimer: I am using the stupidest, most cutting edge setup which may or may not totally break this (Using vscode-remote with WSL on Windows, and passing through COM1 to /dev/ttyS1).
 
I have a similar issue: Before adding ```"port": "/dev/ttyS1"```, neither port selection nor the serial monitor worked and I couldn't upload anything to the MKR1000 on COM1.
Triggering either ```Select Serial Port``` or ```Open Serial Monitor``` terminated the extension host and even made me restart VSCode with WSL totally to regain any functionality.
They still do, but since adding the port I still can't open the serial monitor. 
Workaround for this is to ```cat /dev/ttyS1``` either in the built in or the external terminal, but having this functionality from the extension would be way better.

I know this is a quite loaded issue, and it reeks of preview software. Still I'd be grateful for any help!

bmo-at, 2019-05-17:
@C0rn3j I don't know if this helps you, but instead of relying on the Arduino Extension you can try to set up some tasks in .vscode/tasks.json

I cobbeled [this][1] together yesterday to help alleviate most of my issues with the Arduino Extension like crashes when selecting a port.

[1]: https://gist.github.com/bmo-at/5a70df1b9f12df08a66bb9798eae97b0

zunhu, 2019-05-29:
@C0rn3j  I've tried to launch VScode and directly press F1 to select "Arduino: Select Serial Port", it will show error like your description.
But seems it is usage steps different.
May I know your previous steps? Have you created project or open any example? Because when I open an example and click <Select Serial Port> it can be select by normal.

C0rn3j, 2019-05-29:
1. connect my ESP8266 board so /dev/ttyUSB0 exists
2. create new folder /tmp/ard and open it in VSC
3. create new file /tmp/ard/something.ino and open it in VSC
4. create /tmp/ard/.vscode/arduino.json because selecting port is broken
```
{
	"port": "/dev/ttyUSB0"
}
```
5. Click "Open serial monitor" icon
[image]

zunhu, 2019-06-11:
@C0rn3j I just got the ESP8266 device.

**for your step** 4, I don't think it is necessary, you can skip step4 and select serial port, arduino.json file will be generate automatically.

my steps:
1,connect my ESP8266 board
2,create new folder and use VSC to load folder.
3,create new file and open it.
4,select serial port.

my test environment:
Ubuntu 18.04
VSCode 1.35.0
Arudino IDE 1.8.9

C0rn3j, 2019-06-14:
Could you link the commit which fixes this?

zunhu, 2019-06-17:
@C0rn3j I got same error in manjaro. Bug reopened. Let me do more investigation and will update you later.
BTW, when you use ArduinoIDE(not in VSC), to do select port, upload, e.g., is everything works well?

C0rn3j, 2019-06-17:
Yes, Arduino IDE works without a hitch.

lackdaz, 2019-07-05:
this is definitely still present, I was working on vscode-arduino about 2-3 weeks back without a hitch but I can't select my serial port anymore. Now its either I upload via Arduino IDE or manually input the port into the `arduino.json`

MacOSX 10.14.5
VSCode: 1.36.0
vscode-arduino: 0.2.26

br1pro, 2019-07-05:
Hi! The same issue here. Can you explain better if some workaround are available?
Can I modify the arduino.json file manually?

Mrcuve0, 2019-07-05:
> I do not get any feedback however (no errors) when I hit Select Serial Port. Just nothing

Can confirm this behavior. At the same time I cannot open the serial monitor on VSCode which is giving me the exact same error depicted in the very first post.
Also, like the first post suggested, I added manually the serialPort into the ```arduino.json``` file and at least I'm able to upload the sketch.

I tried opening the Serial Monitor using the Arduino IDE and it opens, but I don't receive any message from the board. (might be a problem on my side probably, not linked with this issue.)
My user has all the permissions to access ```/dev/ttyUSB0``` which is the port where my ESP is connected.

@br1pro As a workaround you can open your terminal and type ```screen /dev/ttyUSB0 115200```.
Change the Baud rate accordingly to your settings of course.
You may have to reset your board once typed the command. It's not the best solution but at least you can open your serial monitor now.

Here's some infos:
```
Version: 1.36.0
Arduino Version: 1.8.9
Commit: 0f3794b38477eea13fb47fbe15a42798e6129338
Date: 2019-07-03T13:20:56.241Z
Electron: 4.2.5
Chrome: 69.0.3497.128
Node.js: 10.11.0
V8: 6.9.427.31-electron.0
OS: Linux x64 4.19.56-1-MANJARO
```

DanwithArduino, 2019-07-09:
Same error here. Cannot change or open Serial/COM port in VScode

sreebhargava143, 2019-11-12:
VSCode serial port issue is not there till may2019 update it started from june 2019 update so to solve the serial port issue with arduino extension I rolled back to the [version released on may2019](https://code.visualstudio.com/updates/v1_35) it is working fine. Remember to make the **"update.mode" : "none"** in settings.

hellyzh, 2020-01-06:
Is the issue still exist? I am alse not able to reproduce it if use the latest version of Arduino extension and VSCode.

jerver, 2020-01-11:
Issue still exists on my installation.

hellyzh, 2020-01-13:
@jerver , have you used the latest Arduino extension 0.2.29 and vscode 1.41.1? And what is your OS version? Is the serial port working properly using Arduino IDE? Because I could not repro the issue, please give more details.

sreebhargava143, 2020-01-13:
Yeah port is working correctly on Arduino ide but vscode Arduino extension has some issues.
I used latest vscode and arduino extension but no use.
I'm using windows 10 Home basic.

hellyzh, 2020-01-13:
@sreebhargava143, I just tested on windows 10 home x64 edition, the serial port can open and run use Arduino extension. Do you installed the correct hardware USB driver like [ST-Link/V2](https://www.st.com/en/development-tools/st-link-v2.html)?

hellyzh, 2020-02-17:
Close the issue because it doesn't have recent activity. Please feel free to open a new issue if you would like further discussion. Thank you for your contributions.

closed 2020-02-17 by hellyzh

wouterds, 2020-03-10:
Still happens

ro4stbeef, 2020-03-11:
The issue still exists.

szohar, 2020-03-11:
Any luck here, I am pretty much stack now. It used to work.

wouterds, 2020-03-11:
> Any luck here, I am pretty much stack now. It used to work.

Turn automatic updates off or set to manual and downgrade to [January 2020 release](https://code.visualstudio.com/updates/v1_42).

luhua2018, 2020-03-13:
问题依然存在，我重装了几次VSCode也没有解决，正在考虑换一个低版本的。

luhua2018, 2020-03-13:
> 1. 连接我的ESP8266开发板，以便/ dev / ttyUSB0存在
> 2. 创建新文件夹/ tmp / ard并在VSC中打开
> 3. 创建新文件/tmp/ard/something.ino并在VSC中打开它
> 4. 创建/tmp/ard/.vscode/arduino.json，因为选择端口已损坏
> 
> ```
> {
> 	"port": "/dev/ttyUSB0"
> }
> ```
> 
> 1. 单击“打开串行监视器”图标
>    [image]

2020年3月，我遇到了同样的问题

JinhangZhu, 2020-03-16:
- Arduino 0.2.29
- VSCode 1.43.0

The bug still exists here in my Windows: `Clicking "Select a Serial Port" does noting`.

chris-schmitz, 2020-03-18:
- VSCode 1.43.1
- Arduino 0.2.29
- Arduino IDE 1.8.10

The bug definitely still exists for me as well. Clicking serial port in the lower menu does nothing. Trying to select the serial port from the command pallet also does nothing. 

If I manually add the port in my `.vscode/arduino.json` file I can still upload, but not serial monitoring. 

And all of this works fine in my arduino IDE. 

[image]

kingcoyote, 2020-03-20:
Still happening to me with VSCode 1.43.1 and Arduino 0.2.29 on Windows 10.

When I click on the select serial port link, nothing happens. When I manually enter "COM3" into the arduino.json file, I can upload to my ESP32 just fine.

It works fine on my work computer, but not from my home computer (yay COVID-19 shifting my work location).

One thing of note - on my work computer, I had Arduino IDE installed before I installed VSCode and the Arduino extension. On my home computer, I installed VSCode and Arduino extension first, had errors with arduino.path that made me realize I never installed Arduino IDE here, then installed it and hit this issue with the serial port. Maybe the bug only happens if you install those 3 things out of order?

Good: Arduino IDE -> VSCode -> Arduino Extension
Bad: VSCode -> Arduino Extension -> Arduino IDE

Rafdal, 2020-03-21:
Still happening to me too, I have the lastest version of VSCode, Arduino and Ubuntu Linux

Ben-Chapman, 2020-03-23:
This is also an issue for me with the 1.43.1 version.

Rolling back to 1.42.1 fixed my issues with the port selection and Serial Monitor. For reference, if on Debian/Ubuntu-based systems: `apt install code=1.42.1-1581432938`

mich2k, 2020-06-01:
Still having this bug, unluckyly the extension is not usable

Ephrem-Adugna, 2020-06-18:
Had the same problem, fixed it by unpluging a microphone that was plugged in at the time

MitchTalmadge, 2020-06-27:
Still an issue

Zamtakk, 2020-07-31:
I had this problem until I updated to the latest version of Visual Studio Code Insiders. Select Serial Port works fine now.

Version: 1.48.0-insider (user setup)
Commit: c9a2f78283b6e5ef708fb8869e2a5adaa476e42f
Date: 2020-07-31T05:52:44.330Z
Electron: 7.3.2
Chrome: 78.0.3904.130
Node.js: 12.8.1
V8: 7.8.279.23-electron.0
OS: Windows_NT x64 10.0.18362

Arduino extention version: v0.3.1

Arduino IDE version 1.8.12

mtorromeo, 2020-09-12:
I fixed the problem without switching to the vscode insiders build.
The extension has a couple of modules that have native bindings built for a different version of node/electron.

1. Install electron-rebuild. I use yarn: `yarn global add electron-rebuild`.

2. `cd $HOME/.vscode-oss/extensions/vsciot-vscode.vscode-arduino-0.3.2/out/node_modules/usb-detection` (change the .vscode-oss path and the version of vscode-arduino accordingly to your environment)

3. `yarn add nan; electron-rebuild -v 9` (9 is the version of electron used by your vscode)

I configured node to install global packages in `$HOME/.npm-packages` with `$HOME/.npm-packages/bin` in my PATH. You have to adapt the call to `electron-rebuild` based on your environment.

4. `cd $HOME/.vscode-oss/extensions/vsciot-vscode.vscode-arduino-0.3.2/out/node_modules/@serialport/bindings; yarn add nan; electron-rebuild -v 9`

The serial port selection and monitor should work fine now.

semi921, 2020-09-15:
@mtorromeo's method solved my problem. Now it's working!

fregas, 2020-09-20:
I too have the same issue on windows 10.  Had the latest version of vscode 1.49.0.  I had to uninstall vscode, downgrade to the 1.42.1 version and uninstall the C++ and arduino plugins, then reinstall arduino which then gets the correct working version of C/C++ plugin its dependent on.

lesmalave, 2020-09-20:
Great, it worked for me 🙂👍

lesmalave, 2020-09-20:
> I fixed the problem without switching to the vscode insiders build.
> The extension has a couple of modules that have native bindings built for a different version of node/electron.
> 
> 1. Install electron-rebuild. I use yarn: `yarn global add electron-rebuild`.
> 2. `cd $HOME/.vscode-oss/extensions/vsciot-vscode.vscode-arduino-0.3.2/out/node_modules/usb-detection` (change the .vscode-oss path and the version of vscode-arduino accordingly to your environment)
> 3. `yarn add nan; electron-rebuild -v 9` (9 is the version of electron used by your vscode)
> 
> I configured node to install global packages in `$HOME/.npm-packages` with `$HOME/.npm-packages/bin` in my PATH. You have to adapt the call to `electron-rebuild` based on your environment.
> 
> 1. `cd $HOME/.vscode-oss/extensions/vsciot-vscode.vscode-arduino-0.3.2/out/node_modules/@serialport/bindings; yarn add nan; electron-rebuild -v 9`
> 
> The serial port selection and monitor should work fine now.

Great, it worked for me 🙂👍

simonl65, 2020-09-20:
As a workaround:

1. open the folder where your .ino file resides
2.  open the .vscode folder
3.  edit arduino.json
4. change the port
    e.g.
```
{
  "port":"COM4"
}
```

tuxedo0801, 2020-09-27:
@hellyzh Why you're closing this if the issue is still present?!

Balzermw, 2020-09-29:
> As a workaround:
> 
> 1. open the folder where your .ino file resides
> 2. open the .vscode folder
> 3. edit arduino.json
> 4. change the port
>    e.g.
> 
> ```
> {
>   "port":"COM4"
> }
> ```

Worked for me as a workaround. 

Still broken on version 1.49.2

ShrirajHegde, 2020-10-02:
> I fixed the problem without switching to the vscode insiders build.
> The extension has a couple of modules that have native bindings built for a different version of node/electron.
>
> 1. Install [...]

> I fixed the problem without switching to the vscode insiders build.
> The extension has a couple of modules that have native bindings built for a different version of node/electron.
>
> 1. Install [...]

Works Great! , is there any way you can fix the bug permanently without this patch required?

ShrirajHegde, 2020-10-02:
> Great, it worked for me

Works Great! , is there any way you can fix the bug permanently without this patch required?

Andreslion, 2020-10-11:
> I fixed the problem without switching to the vscode insiders build.
> The extension has a couple of modules that have native bindings built for a different version of node/electron.
>
> 1. Install [...]

How do I do this in windows? :(

ShrirajHegde, 2020-10-12:
> How do I do this in windows? :(

Install yarn, install electron-rebuild then maybe use "everything" from voidtools to search the executables and add them to PATH (if they are not)
Then search .vscode folder in c:\users\<user> directory and follow the same steps

again easier thing to use is "everything", just search .vscode and select the right one if you get multiple

Andreslion, 2020-10-12:
> Install yarn, install electron-rebuild then maybe use "everything" from voidtools to search the executables and add them to PATH (if they are not)
> Then search .vscode folder in c:\users<user> [...]

Thanks! I'll give it a try!

JJFourie, 2020-10-14:
So from May 2019 until October 2020 the problem can't be fixed?
Time to move on to some other app??

ykchau, 2021-02-06:
I just find this maybe the order of statement problem, I put the "port" setting at the 2nd row of the arduino.json, and restart VS Code, it works, however another problem arise (Failed to open serial port COM6 due to error), looks like this problem is from VS Code itself, I can still compile and upload the program via that serial port, just can't open it with the plug icon at the bottom right corner of VS Code.

{
    "sketch": "sketch.ino",
    "port": "COM6",
    "board": "esp8266:esp8266:d1_mini",
    "configuration": "xtal=80,vt=flash,exception=legacy,ssl=all,eesz=4M2M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600"
}

0xStarcat, 2021-02-07:
Founds because the issue is still present using version `0.3.4` vscode extension on ubuntu linux 

the workaround editing `.vscode/arduino.json` works but it'd be preferable not to have to do this for every project.

RedaBousbah, 2021-02-08:
Last VS Code update (January 1.53) breaks again this functionality (at least on Linux). Had to downgrade back to 1.52 to get it to work.

Rafdal, 2021-02-08:
To be honest, I couldn't solve this problem.

Uninstalled the Arduino extension and installed PlatformIO. 

Maybe it's tricky at the first time but then when you got the hang, it always works like a charm, never had a problem in almost 2 years of use.

You can do everything from PlatformIO, install Libraries, use different platforms and boards (ESP32, Esp8266, Arm boards, Heltec Lora 32, all the Arduino family, and +1000 more.

Think about it ;)

ashwinexe, 2021-08-21:
> I fixed the problem without switching to the vscode insiders build.
> The extension has a couple of modules that have native bindings built for a different version of node/electron.
>
> 1. Install [...]

I am getting invalid electron version, even after entering the latest (v13) throws the same error