microsoft/vscode-arduino #818: warning with default includePath after initialize, `cannot open source file "avr/pgmspace.h"`

bbukaty, 2019-05-16:
when I did a normal initialize in a new folder just now and made a default ino file, I immediately got this error:
`cannot open source file "avr/pgmspace.h" (dependency of "C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino\Arduino.h")`
well, warning not error. nevertheless, to fix it I had to add this to includePath:
`"C:\\\\Program Files (x86)\\\\Arduino\\\\hardware\\\\tools\\\\avr\\\\avr**"`
If a default ino script with no additional libraries or anything is giving me this warning because that wasn't in includePath, should that not just be in includePath by default?
I'm on the latest version, `0.2.25`.

ShixiangWang, 2019-06-13:
same issue in linux

closed 2019-07-18 by stale[bot]

BasixKOR, 2019-07-30:
Having same issue.

schupat, 2019-10-20:
Same issue under Ubuntu 18.04

I searched for the file using `find -name 'pgmspace.h'` from within my Arduino installation folder and found the file there. Was required to add `"/home/[your_username]/.bin/arduino-1.8.10/hardware/tools/avr/avr/**"` to my `c_cpp_properties.json` file under `includePath`. **Adding the full path doesn't work.**

Change the `/home/[your_username]/.bin/arduino-1.8.10/` part to wherever you have installed your Arduino IDE and you should be fine.

lguilhot, 2020-04-10:
thank you schupat,
for me it worked with "..hardware/tools/avr/**" and not "...hardware/tools/avr/avr/**"

ah3243, 2020-05-10:
> thank you schupat,
> for me it worked with `"..hardware/tools/avr/**"` and not `"...hardware/tools/avr/avr/**"`

could you quote the absolute path, it's still throwing up errors

justummar, 2020-05-31:
I get the same thing in windows? I added to `settings.json` file which has this in it
`"arduino.defaultBaudRate": 9600,
    "arduino.logLevel": "verbose",
    "arduino.path": "C:\\Program Files (x86)\\Arduino",
    "C_Cpp.default.includePath": ["C:\\Program Files (x86)\\Arduino", 
                                  "C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr\\avr\\include\\avr\\",
                                  "C:\\Program Files (x86)\\Arduino\\hardware\tools\\avr\\avr\\include"
                                ],`
I added the C_cpp.default stuff but still get the squiggly lines...

brunosacom, 2020-09-05:
I solve it adding the follow line in the file: c_cpp_properties.json
"includePath":[
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\**"
]