microsoft/vscode-arduino #820: Exit with code = EPERM

MHz000, 2019-05-17:
Issue Type: <b>Bug</b>

checking or compiling/download an Arduino.ino file always resutls in
Exit with code = EPERM
using VSc 1.34.20

Extension version: 0.2.25
VS Code version: Code - Insiders 1.34.20-insider (e83e24a0c9225becd7341e56952177a20d5d4629, 2019-05-15T18:06:45.467Z)
OS version: Windows_NT x64 10.0.17763

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i5-7260U CPU @ 2.20GHz (4 x 2208)|
|GPU Status|2d_canvas: enabled<br>checker_imaging: disabled_off<br>flash_3d: enabled<br>flash_stage3d: enabled<br>flash_stage3d_baseline: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>native_gpu_memory_buffers: disabled_software<br>rasterization: enabled<br>surface_synchronization: enabled_on<br>video_decode: enabled<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|undefined|
|Memory (System)|7.89GB (3.94GB free)|
|Process Argv||
|Screen Reader|no|
|VM|0%|
</details>

ArthurMa1978, 2019-05-28:
which board did you use?

MHz000, 2019-05-28:
I'm using WeMos  D1 R2 and LOLIN 32. Tried complet reinstallation of
microsofts vscode for arduino, no success.
Maybe I missed one of the many settings. Deploying latest arduino IDE 64
bit WIN10.

Thank you for your advice!

closed 2019-07-02 by stale[bot]