microsoft/vscode-arduino #828: [JLink] Default LaunchFile missing attributes to connect

ooeygui, 2019-06-01:
When creating a new Arduino launch file, the attributes are missing entries which are required by the MIDebugger to connect and find the server start command.

The "serverStarted" entry is used by the MI Debugger to determine when the server has been started. However, the output can only be found if you also set filterStdout - since "Connected to target" is only sent on stdout.

            "serverStarted": "Connected to target",
            "launchCompleteCommand": "exec-continue",
            "filterStderr": true,
            "filterStdout": true,

**Action:**
Please add filterStdout to the standard launch file - or infer it when serverStarted is non-null.

pieandcakes, 2019-06-01:
Another option would be to validate that they are true and if they are not, output a warning before calling the C++ extensions

ArthurMa1978, 2019-06-03:
https://github.com/microsoft/vscode-arduino/commit/6d7ed7d75ab931852ca648df66067607d25f5059

will add filterStdout:true

closed 2019-07-08 by stale[bot]