microsoft/vscode-arduino #829: Arduino.h and ESP8266 includes have squiggles

supaplextor, 2019-06-04:
Ubuntu 18.04.2 LTS
An hourly build of Arduino 1.8.10 from about May 25-30th (rock solid on its own)
```

{
    "configurations": [
        {
            "name": "Linux",
            "includePath": [
                "${workspaceFolder}/**",
                "${HOME}/Arduino/libraries/**"
            ],
            "defines": [],
            "compilerPath": "/usr/bin/gcc",
            "cStandard": "c11",
            "cppStandard": "c++17",
            "intelliSenseMode": "clang-x64"
        }
    ],
    "version": 4
}

```

[image]

[image]

[image]

Not sure what I did since yesterday, but I did have most of it working except these two libraries to include:

https://github.com/nickgammon/bitBangedSPI
https://github.com/nickgammon/MAX7219_Dot_Matrix

seweis, 2019-06-05:
Here are a few pointers, someone helped me with mine. 

1. Add the full path in the includepath. 
2. the compiler path, to include the full path. 
3. Then after that.  Make sure the .c_cpp_properties.json file is inside the .vscode folder.
4. And finally, the workspace to only have one file c_cpp_properties.json
5.the verison i am using is C++ intellisense 0.32.1

supaplextor, 2019-06-05:
Not sure what you mean on item 2.
Check on 1, 3, 5 and 5. I'm using the same version.

```
{
    "configurations": [
        {
            "name": "Linux",
            "includePath": [
                "${workspaceFolder}/**",
                "${HOME}/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/**",
                "${HOME}/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/**",
                "${HOME}/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/include/**",
                "${HOME}/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/lwip2/include/**",
                "${HOME}/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tests/host/**",
                "${HOME}/Arduino/libraries/bitBangedSPI-master/**",
                "${HOME}/Arduino/libraries/MAX7219_Dot_Matrix-master/src/**",
                "${HOME}/Arduino/libraries/DHT_sensor_library_for_ESPx/**"
            ],
            "defines": [],
            "compilerPath": "/usr/bin/gcc",
            "cStandard": "c11",
            "cppStandard": "c++17",
            "intelliSenseMode": "clang-x64"
        }
    ],
    "version": 4
}
```
```

> Executing task in folder LED_matrix: /usr/bin/g++ -g /home/supaplex/Downloads/LED_matrix/led-matrix-vs-vscode.cpp -o /home/supaplex/Downloads/LED_matrix/led-matrix-vs-vscode <

/home/supaplex/Downloads/LED_matrix/led-matrix-vs-vscode.cpp:1:10: fatal error: Arduino.h: No such file or directory
 #include <Arduino.h>
          ^~~~~~~~~~~
compilation terminated.
The terminal process terminated with exit code: 1

```
[image]

[image]

[image]

I can control click on Arduino.h to see the copy. 

Looks like I have some other housekeeping to do. platformio is disabled right now.

```
supaplex@supaplex-gtx760m:~$ find / -name Arduino.h -ls 2>/dev/null 
 43129352      8 -rw-r--r--   1 root     root         5844 Dec  8  2016 /usr/share/arduino/hardware/arduino/cores/arduino/Arduino.h
 43129392      8 -rw-r--r--   1 root     root         5844 Dec  8  2016 /usr/share/arduino/hardware/arduino/cores/robot/Arduino.h
  6970453      8 -rw-rw-r--   1 supaplex supaplex     7483 Sep 10  2018 /home/supaplex/usr/src/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino/Arduino.h
  7609587     12 -rw-r--r--   1 supaplex supaplex     9141 May 20 11:56 /home/supaplex/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tests/host/common/Arduino.h
  7609460     12 -rw-r--r--   1 supaplex supaplex     9528 May 20 11:56 /home/supaplex/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/Arduino.h
   395832     12 -rw-r--r--   1 supaplex supaplex     9141 May 20 11:56 /home/supaplex/.platformio/packages/framework-arduinoespressif8266/tests/host/common/Arduino.h
   397484     12 -rw-r--r--   1 supaplex supaplex     9528 May 20 11:56 /home/supaplex/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Arduino.h

```

seweis, 2019-06-05:
Search for the "arduino.h" and inculd that path. 

For item 2. The compiler us have usr/.. i guess if you are using linux thats were it would be , which might be correct.

supaplextor, 2019-06-06:
Terse sample code at https://gist.github.com/supaplextor/1f1b32ba9849a0144ae3d426a1f9510c to reproduce the same issue.  Builds , uploads and runs okay in Arduino 1.8.10.  Should I retry with 1.8.8?
Board is a nodemcu v1 clone running ESP8266EX.

vscode has squiggles for WProgram.h in /home/supaplex/Arduino/libraries/DHT_sensor_library_for_ESPx/DHTesp.h.  Based on line 46 there it looks like ARDUINO is one reason why.  Depending on how I edit .vscode/c_cpp_properties.json I get up to three references for Arduino.h.

```

#if ARDUINO < 100
  #include <WProgram.h>
#else
  #include <Arduino.h> // greyed out
#endif

```

supaplextor, 2019-06-06:
Current c_cpp_properties.json
```
{
    "configurations": [
        {
            "name": "Linux",
            "includePath": [
                "${workspaceFolder}/**",
                "${HOME}/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/**",
                "${HOME}/Arduino/libraries/bitBangedSPI-master/**",
                "${HOME}/Arduino/libraries/MAX7219_Dot_Matrix-master/src/**",
                "${HOME}/Arduino/libraries/DHT_sensor_library_for_ESPx/**"
            ],
            "defines": [],
            "compilerPath": "/usr/bin/gcc",
            "cStandard": "c11",
            "cppStandard": "c++17",
            "intelliSenseMode": "clang-x64"
        }
    ],
    "version": 4
}
```

supaplextor, 2019-06-15:
If I define ARDUINO just as the shell is using during build, it will build, and no squiggles on header files now. Ugly hack :) ...

```
#define ARDUINO 10808
#include <Arduino.h>
#include <DHTesp.h>

DHTesp dht;

void setup() {
  Serial.begin(115200);
  dht.setup(D5, DHTesp::DHT11);
}

float temp = 0;
float ltemp = 1;

void loop() {
  delay(25);
  temp = dht.getTemperature();

  if ((!isnan(temp)) && (ltemp != temp)) {
    Serial.printf("%0.1f F\n", dht.toFahrenheit(temp));
    Serial.printf("%0.1f humidity\n", dht.getHumidity());
    if (temp != ltemp) {
      ltemp = temp;
    }
  }
}

/* Sample Serial output (with timestamps enabled on monitor)

10:11:05.541 -> 80.6 F
10:11:05.541 -> 16.0 humidity
10:19:00.347 -> 78.8 F
10:19:00.347 -> 16.0 humidity

*/
```

[image]

samigroup, 2019-06-15:
I see... What a hack!!! 

Q: Did you have to add the Arduino extension?

supaplextor, 2019-06-15:
It's installed. (tossing some screenshots for the next guy)

[image]

Had to adjust the Arduino path. I cleaned up the hourly build of Arduino, installed 1.8.8 to /usr/local/bin/arduino-1.8.8. The hourly build was rock solid, but I moved back to go along with the vscode install instructions.
 
[image]

User Settings (hit F1)
[image]

closed 2019-07-20 by stale[bot]

muesli, 2019-07-29:
I can still reproduce this with the latest release of both the app and the extension on Arch Linux.

elektronikworkshop, 2020-02-18:
This is due to the improper IntelliSense implementation of vscode-arduino. I'm currently working on this and a beta should be available soon at least for OSX and Linux. Windows will follow.

Anyone interested can track the progress [here](https://github.com/elektronikworkshop/vscode-arduino/blob/intellisense-autoconfig/BRANCHNOTES.md) and I often report to issue #438 as well.

Regards
EW

phlipse, 2020-12-19:
Any news about this? I still can reproduce it with the latest version.

adiazulay, 2021-01-05:
this should be resolved once #1141 is merged. I expect it to be later this week.