microsoft/vscode-arduino #830: How to add additional library folders?

jovabel, 2019-06-05:
arduino-builder supports the "-libraries" parameter to add additional library folders (multiple!).

Both arduino.json and settings.json  (in the workspace/.vscode/* folder) do not support setting the "-libraries" parameter,  thus making it impossible to put code that is used in multiple sketches at other locations than the default library location ("C:\Program Files (x86)\Arduino\libraries") and the user library location ("C:\Users\<USER>\Documents\Arduino\libraries").

This is very inconvenient for development environments that do not allow the use of these local directories. For example: environments which demand all code (sketches and shared code) to be under source control. 

Please add a setting for adding additional library folders (enable the use of the -libraries parameter).

The same holds for being able to customize the 'sketchbook location' in arduino.json and settings.json.

jovabel, 2019-06-28:
Not having the option to add additional library folders makes the vscode-arduino extension impractical (an not usable) in a more professional sw development environment.

mMerlin, 2019-07-22:
A partial work around (currently) is to change the `sketchbook.path` entry in the Arduino preferences.txt file.  `$HOME/.arduino15/preferences.txt` on my Fedora 30 system.
A single entry, but at least you can move the sketchbook, and contained libraries path, to some place else.  Per user, per folder, configuration overrides, and a list, would be nice.

closed 2019-08-26 by stale[bot]

Gueka, 2019-09-18:
Revive this issue.
Excellent editor is worthless if we have to change to another editor to include libraries.

jovabel, 2019-09-25:
> 
> 
> Revive this issue.
> Excellent editor is worthless if we have to change to another editor to include libraries.

I agree fully. Also The effort to implement should be is small, because arduino-builder supports this by using the -libraries parameter. So the issue is purely: make it configurable in arduino.json or settings.json and then use the configured values with the -libraries parameter when calling the build executable (arduino-builder). The trick of using values from arduino.json or settings.json as parameter for the build process is already done for other configurable options.... so implementing it for the libraries should be really easy.

iqapps, 2020-05-27:
I found a way around this, but I think its weird someone has made the efford to configure include paths settings, and then limited those to not work with users library paths ?"?¤%"?

The solution (for Windows 10 in my case) is to have a private library somewhere, lets say as a sibling to your project folders, maybe all at the same level, and then make a junction link in the workspace path, to the real library folder. You should then gitignore the junction folder, and add the real folder to Git instead. You can include classes from sub folders by #include "Library/Classes/someclass.h", provided the junction is named "Library". I use this setup for many projects with VS code and the MS Arduino extension, having many private classes in the Library folder. Its not perfect, but it works and is practical as the "Library" junction and its sub folders and classes are all visible from the project workspace.

stepheneb, 2021-02-05:
I have a project which includes a `./libraries` folder with a number of included libraries which works fine with the Arduino IDE 1.8.13 when I set my **Sketchbook location** to the project folder.

I can't seem to figure out any way to do this using the Arduino extension in vscode.

```
$ tree -d libraries
libraries
|-- Adafruit_BusIO
|   `-- examples
|       |-- i2c_address_detect
|       |-- i2c_readwrite
|       |-- i2c_registers
|       |-- i2corspi_register
|       |-- spi_modetest
|       |-- spi_readwrite
|       |-- spi_register_bits
|       `-- spi_registers
|-- Adafruit_MCP4725
|   `-- examples
|       |-- sinewave
|       `-- trianglewave
|-- Adafruit_MCP9600_Library
|   |-- assets
|   `-- examples
|       `-- mcp9600_test
|-- Adafruit_RGB_LCD_Shield_Library
|   |-- examples
|   |   `-- HelloWorld
|   `-- utility
|-- SerialMenu
|   |-- examples
|   |   |-- demo1
|   |   `-- demo2
|   `-- src
`-- arduino-sht
    `-- examples
        |-- multiple-sht-sensors
        |-- sht-autodetect
        `-- sht3xanalog
```

vscode:
```
Version: 1.53.0
Commit: 8490d3dde47c57ba65ec40dd192d014fd2113496
Date: 2021-02-03T16:01:04.046Z (1 day ago)
Electron: 11.2.1
Chrome: 87.0.4280.141
Node.js: 12.18.3
V8: 8.7.220.31-electron.0
OS: Darwin x64 19.6.0
```

vsciot-vscode.vscode-arduino: v0.3.4

fkromer, 2021-08-04:
I use a per project `./libraries` folder as well. Manually copy & paste the libs from the global libary path into the per project path is error prone w.r.t. library versions, etc.

i0n-, 2021-08-23:
My workaround was add my local folder through Arduino IDE:

[image]

[image]

i0n-, 2021-08-23:
> My workaround was add my local folder through Arduino IDE:
> 
> [image]
> 
> [image]

Only work once... Changes on library is not updated   
:-(

i0n-, 2021-08-23:
FYI: https://www.arduino.cc/en/Guide/Libraries

Finally I found a working workaround...

I install this extension:
https://marketplace.visualstudio.com/items?itemName=pavel-purma.copy-watcher

Then in my xxx.code-workspace file I configure the extension, like this:
```
{
	"folders": [
		{
			"path": "."
		},
		{
			"path": "..\\Tst-Lib"
		}
	],
	"settings": {
		"copyWatcher.sections": [			
			{
				"source": "..\\Tst-Lib",
				"destination": "xxx\\Documents\\Arduino\\libraries\\Tst-Lib",
                                "destinationRequired": false,
				"deleteEnabled": true,
				"initialCopy": true
			}

		]
	}
}
```
"destination" must be the same path as `sketchbook.path` propety - "Sketchbook location:"

[image]

Then restar the VSCode windows and start to work the library, with the new changes ready.

i0n-, 2021-08-23:
Forget everything....

At the end I work my libraries inside the Sketchbook location and added the folder to the workspace.

 😒

grandehombre, 2021-12-11:
I simply create a './libraries' folder in my workspace and drop shortcuts in it of the all library folders I need.
That way the compiler finds what it needs, yet I get to keep a single copy of the libraries, which helps with updates

This works with Win10 and VSCode v1.63.0

I hope this helps...

grandehombre, 2021-12-11:
I simply create a './libraries' folder in my workspace and drop  symbolic links (*1) in it of the all library folders I need.
That way the compiler finds what it needs, yet I get to keep a single copy of the libraries, which helps with updates

This works with Win10 and VSCode v1.63.0

I hope this helps...

*1: Note that I use the "HardLinkShellExt_X64" windows explorer extension to create symbolic links in the libraries folder.
You can find it [here]( https://www.techspot.com/downloads/7107-link-shellextension.html)

phed, 2022-07-27:
A simple workaround I found. Maybe this works differently with arduino-cli, but at heart arduino uses "arduino_debug.exe" which again dispatches "arduino-builder.exe". This is what at heart limits the ability to add multiple library paths.

> arduino.json
```
{
    ..., 
    "buildPreferences": [
        ["sketchbook.path", "../boat/lib/"]
    ]
}
```
This will set the new search path to "../boat/lib/libraries/" relative to the ino-file. Using "${workspaceFolder} does not work. Note that it might create a readme.txt if the folder doesn't already exist.

jleaders, 2023-07-23:
Alternatives would be an arduino cmake toolchain, but they are really hard to setup

so I used Windows built in command [mklink](https://learn.microsoft.com/en-us/windows-server/administration/windows-commands/mklink) for an NTFS junction (symlink), this fools the file system into believing a folder is synchronized in two places without taking up any more space or requiring any actual syncing

In integrated cmd terminal in VS Code:
`mklink /J myinopath/folder libraries/mylib/folder`
Where the link goes alongside the *.ino in a subdirectory
Then 
`#include "mylib/folder/file.h"` works

Caveats: 
* But this won't support files like *.inl that don't end in typical C++ extensions
* This must be done in cmd, not powershell
* Ignore this folder in your source control
* This works in macos/linux as well with `ln -s`
* Consider adding this to a bootstrapping python script to automate this later, I use [Organic Projects](https://leadersandco.com/organic) to automate project flow

bereldhuin, 2024-01-06:
+1,

Please add the additional library path.