microsoft/vscode-arduino #833: Allow C_Cpp.intelliSenseEngine to be set to "Default" instead of "Tag Parser" for better code completion/suggestions

Rossosaurus, 2019-06-16:
This is extension is great and all but it would be nice if the intellisense engine could be set to default so it doesn't show every function from every library every time you type something. I don 't know if it's a bug or something that was done on purpose but it's something that I see a lot of posts and content on but never a solution.

[Here is a post I made a couple of days ago about the problem](https://www.reddit.com/r/vscode/comments/c0xi3o/c_arduino_extension_intellisense_wont_work_unless/) and [here's another issue about tag parser causing issue's](https://github.com/microsoft/vscode-arduino/issues/831). In my opinion it would improve quality of life while using this extension massively.

Thanks

elektronikworkshop, 2020-02-05:
Hi to everyone who is suffering from this!

I was so annoyed, that I began to work on an auto-configuration mechanism, which parses the build output from Arduino and creates a `c_cpp_properties.json` from it. The goal is that no one wants likely ever edit this file manually again - except for very weird setups for which I will add a configuration flag/option that will keep the extension from touching the file.

I coded some proof of concept already but I haven't committed anything yet. But I began documenting the project [here](https://github.com/elektronikworkshop/vscode-arduino/blob/intellisense-autoconfig/BRANCHNOTES.md).

Anyone who wants to keep me motivated can chip in some :beer:-money - the donation button can be found in the link above.

Other support is appreciated and comprises
* people with Windows and OSX setups to test the cross platform compatibility
* the original maintainers could help me with CI and how to build test-releases

The progress can be followed within the same file. The [repo](https://github.com/elektronikworkshop/vscode-arduino) is set to this feature branch by default.

adiazulay, 2020-12-16:
Tracking this with #438

closed 2020-12-16 by adiazulay