microsoft/vscode-arduino #835: [Board Manager] Board variant and other options not presented for some boards

raghur, 2019-06-20:
Arduino IDE - 1.8.9
Boards - STM32 1.6.0 installed via additional urls.
Code for the board can be compiled and flashed with arduino ide.

In VS Code, for the board I have (Blue pill, STM32F103 generic), the dropdown shows the Board - but nothing after that (variant, upload method etc). Re-selecting the board from the drop down does not help for STM32F1 series.

Some other board series also show somewhat similar behavior (STM32F0 series) where selectingg the board does not show other options - but if you click on the same board again in the dropdown, the options show up.

I can compile code in VS but need either CLI or arduino ide to upload. Can you please address this?

raghur, 2019-06-20:
PS - Uninstalling and reinstalling the addon seems to fix this - so probably some cached data?

raghur, 2019-07-20:
> Please feel free to open a new issue if you would like further discussion.

No thanks - that was what this was for :).  There's been no response from the repo owners either acking the issue or otherwise and a bot just closes the issue (so it means that no one will probably look at it). Good luck with getting community participation/feedback.

dilin-MS, 2019-10-29:
Hi, @raghur could you please provide some snapshot of the issue you mentioned?

closed 2019-11-21 by github-actions[bot]