microsoft/vscode-arduino #838: Automatic .pde support opt-out

thugpanda, 2019-06-24:
#793 added generic .pde support. Now the extension is replacing .pde with .ino file extensions. However, it is changing genuine Processing files as well, making working with Processing hard when still using the arduino extension.

A quick fix is the removal of the code added with commit [71d6980](https://github.com/microsoft/vscode-arduino/commit/71d6980f14f8db909d14bfc18e4b1a0ba105ebd7#diff-45327f86d4438556066de133327f4ca2), only removing `".pde"` from `package.json` (line 509) is insufficient. 

It would be nice to have an option in the extension settings for deactivating this feature if necessary.

Sneezry, 2019-06-25:
As ".pde" is also used by Processing, this request makes sense. Would you like to make a PR to fix this issue?

aecanales, 2019-07-05:
Also ran into this problem, would love if it could be fixed. Would try it myself but I've never worked on an extension before.

Dotrar, 2019-07-16:
Submitted a PR for it.

I wasn't even going to add an option to support PDE because it's not the correct filetype ( I also don't know how to do option-based filetypes ) 

Ultimately, do things proper. If arduino examples are in PDE then THEY are wrong; not the tool.

mMerlin, 2019-07-22:
> If arduino examples are in PDE then THEY are wrong; not the tool.

I can't really agree with that.  The Arduino IDE prior to version 1.0 used the .pde extension for all sketches.  That are plenty of sketches still around with that extension, because there has been no need to make any changes to them.  The work just fine as they are.  Removing the ability to use pde files in the vscode Arduino extension would break compatibility with the Arduino IDE.  It should not force conversion to ino either of course.

Dotrar, 2019-07-22:
The only reason why I or thugpanda are against .pde is because other programs want to use PDE and arduino _specifically_ doesn't.

> That are plenty of sketches still around with that extension

No there isn't; Arduino ide AND this tool would convert it to INO since 1.0 - we are now at 1.8.9
any .pde sketch hasn't been touched for years. Arduino hasn't changed because Arduino is a lazy company; the SD card examples have never worked and other people had to re-write the SD library for them; it's surely just a case that someone has to rename the example extensions as well. 

> The Arduino IDE prior to version 1.0 used the .pde extension for all sketches. 

Because it copied from processing, which is a completely separate tool in its own use.

>Removing the ability to use pde files in the vscode Arduino extension would break compatibility with the Arduino IDE

What does this even mean? using pde files in Arduino extension breaks compatibility with Processing. you win some you lose some; however, Arduino themselves have said that ".pde isn't our extension, .ino is" - while Processing still holds onto pde. 

>It should not force conversion to ino either of course. 

Which is why both the ide and extension save as ino whether you like it or not. Completely daft point.

I submitted a PR at #868 which @Sneezry brought up a great compromise:

> A better way to handle this issue I think is to add a new configuration to enable/disable .pde support, and disable this feature by default. Then everyone should be happy.

Which is what I'm planning on working on after work today;

mMerlin, 2019-07-22:
I stand corrected.  My memory said that Arduino IDE offered to convert pde to ino, but did not enforce it.  Probably a change several versions ago that I did not notice.  Having already converted all of my own pde files to ino.  I've been working with this long enough to see all pde files, then a mixture of pde and ino.  Now __SHOULD__ be only ino.  For Arduino.

vishnumaiea, 2019-07-24:
So when are we going to be able to work with Arduino sketches and Processing files together ? IMO, changing the file extension without user confirmation is recklessness for whatever reason someone is doing it for!

mMerlin, 2019-07-24:
Changing pde to ino was initially being done by the Arduino IDE.  That is not a general purpose editor.  The only things that should be opened there are Arduino sketch files, regardless of what the file extension says.  So there should be no conflict changing the extension there.

Visual Studio code **is** a general purpose editor.  The arduino extension was trying to handle anything that the Arduino IDE supported.  So treat pde files as sketches.  The change that appears to be in the works, is to add a configuration flag to tell the Aruduino extension to ignore pde files.  The default being to ignore them.  With that in place, opening a pde file should do whatever the visual studio code editor would normally do.  If there is a processing extension, that should pick it up.  Or manually set the tab to use java.

You should be able to mix Arduino ino and Processing pde files in the editor as soon as that planned pr gets through to your system.  In the mean time, you could always disable the Arduino extension, and fall back to standard C/C++ handling, without any of the benefits the extension provides.

Dotrar, 2019-07-30:
PR is merged! :tada:

NameOfTheDragon, 2019-08-27:
I see the PR is merged but for what it's worth, I don't think an extension should ever autonomously take the decision to rename a file. That's essentially a data-loss situation. It's still happening - has this PR been rolled into a release yet?

Dotrar, 2019-09-18:
it's merged into main: https://github.com/microsoft/vscode-arduino/commit/b5f7361a7b24a616053dd5d542fb70ff51390a9c

and I agree wholeheartedly; unfortunately there's users out there who disagree and fight for the pde over-write; I think having it as a default-off option is by far the best. and at the very least it is only a rename.

closed 2019-10-23 by stale[bot]

robinator8, 2019-10-26:
Is there anyway to disable this? I have .pde files that are actually processing files and every time I open then they're automatically changed .ino files.

Dotrar, 2019-10-26:
Wait until the next release I believe, or you can download and run from source..

Although I made this change 3 months ago, so I don't know how long it will take for release....