microsoft/vscode-arduino #839: Feature Request: Switching between sketches

shac12, 2019-06-27:
Hi, 

Really like the arduino tool. Autocomplete as well as all the other smart features have made this much better to develop software than the default Arduino IDE. One thing that does need changing is the ability to select which .ino file to upload. I have to change the "sketch" value in .vscode/arduino.json every time I want to change sketches that are in the same folder. Being able to quickly upload different .ino files when testing would make the testing and debugging phase go much smoother. 

Kind Regards,
Sharif

Awbmilne, 2020-03-24:
**This would be an awesome feature to add!**

Even just a **Command Pallete option**, so the Sketch-Path input can be hailed using a Keybinding instead of the Status Bar UI.

If the Command Pallete option is easy enough to implement, a **History for the command** (you can use arrow keys to scroll through list of most recent Sketch-Paths) would be super convenient!

_Keybinding for power users 💪_

**Love Note**: Thanks to all the Arduino Extension maintainers, this extension is super powerful and I cannot stress how much time it saves me!

maddogjt, 2020-10-30:
I've put up a PR with this implemented #1128