microsoft/vscode-arduino #840: cannot find Arduino ide

iplayfast, 2019-06-28:
the environment is Linux (mint 19),
vscode is 1.35.1
 
path is /home/chris/bin/arduino
which points to a shell, which cd's to the directory and runs Arduino. (although I've tried point directly to bin as well)

It always says "Cannot find Arduino IDE. Please specify the "arduino.path" in the User Settings. Requires a restart after change."

I can run it from the command line, or by typing the path. 
which arduino shows this path.
/home/chris/bin/arduino

Running out of options, am I being stupid?

ChipTechno, 2019-07-12:
Hello. I have the same problem.
There are 2 config, <code>arduino.commandPath</code> and  <code>arduino.path</code>. The <code>arduino.path</code> is the directory cotains your arduino executable file. The <code>arduino.commandPath</code> is the executable file name.   
In my case <code>arduino.path</code> is <code>/usr/share/arduino</code> and <code>arduino.commandPath</code> is <code>/arduino</code>

ArthurMa1978, 2019-07-15:
Thanks @ChipTechno, hope this can help @iplayfast

closed 2019-07-15 by ArthurMa1978

MatthieuScarset, 2020-04-26:
For those (like me) who installed Arduino IDE through the Snap platform (e.g. "Software" on Ubuntu 18.04), this is how your config should look like in order to works: 

```json
{
  "arduino.path": "/snap/bin",
  "arduino.commandPath": "/arduino",
}
```

baliganikhil, 2020-05-22:
I use a Mac and I have Arduino running from the Downloads folder. So, this is how I got it running. Hope this helps.

```
{
    "arduino.path": "/Users/myUserName/Downloads",
    "arduino.commandPath": "Contents/MacOS/Arduino"
}
```

sc07-git, 2020-05-26:
Thanks @MatthieuScarset

NoOne03, 2020-11-28:
For those who installed Arduino IDE manually *i.e. using Arduino IDE zip file*. Refer to my reply on StackOverflow https://stackoverflow.com/a/65049630/11798494.

mathew-fleisch, 2021-02-14:
After setting the correct location, a restart was necessary for me to get it working

LionelCe, 2021-05-09:
> After setting the correct location, a restart of vscode was necessary for me to get it working
> for me:
> `{ "arduino.path": "/Applications", "arduino.commandPath": "Contents/MacOS/Arduino" }`

Hi. I have the same config (same installation folders) but I'm still having error message about IDE not found.
I tried so many paths and so many tutorials but nothing is working.
Any help ?

idzabdan22, 2021-05-31:
> Hi. I have the same config (same installation folders) but I'm still having error message about IDE not found.
> I tried so many paths and so many tutorials but nothing is working.
> Any help ?

sama issues bro, i use linux manjaro KDE, still stuck in this problem which cannot find arduino path

sintrb, 2021-06-20:
> Hi. I have the same config (same installation folders) but I'm still having error message about IDE not found.
> I tried so many paths and so many tutorials but nothing is working.
> Any help ?

The following is usefully for me:

```
{
    "arduino.path": "/Applications/Arduino 2.app",
    "arduino.commandPath": "Contents/MacOS/Arduino",
}
```

xichen-de, 2021-09-09:
In my case:
on Mac:
```
"arduino.path": "/Applications",
"arduino.commandPath": "Contents/MacOS/Arduino",
```
on Ubuntu:
```
"arduino.path": "/snap/bin",
"arduino.commandPath": "arduino",
```

xNewz, 2022-10-07:
> After setting the correct location, a restart of vscode was necessary for me to get it working for me: `{ "arduino.path": "/Applications", "arduino.commandPath": "Contents/MacOS/Arduino" }`

Thanks it works for me

sergiyd, 2022-11-04:
Does not work any variant from here.

macOS 13.0 (22A380)
Arduino IDE Version: 2.0.1

`ls /Applications/Arduino\ IDE.app/Contents/MacOS`
outputs
`Arduino IDE`
however playing with _arduino.path_ (/Applications, /Applications/Arduino\ IDE.app, /Applications/Arduino IDE.app) _arduino.commandPath_ (Contents/MacOS/Arduino IDE, Contents/MacOS/Arduino\ IDE) didn't bring any success

ericoud, 2022-11-10:
Same issue with macOS 11.7.1 and Arduno V2.0.1
I tried almost every possible syntax, nothing works...

cyclops1982, 2022-11-11:
For anybody with Arduino v2.0.1 and macos 12.6. I use this config:
```
{ "arduino.path": "/Applications/Arduino IDE.app", "arduino.commandPath": "Contents/MacOS/Arduino IDE" }
```

I installed arduino by downloading it.  My vscode is from brew.sh.

cdealti, 2022-11-13:
macOS 12.6
Arduino IDE Version: 2.0.1

```
$ ls /Applications/Arduino\ IDE.app/Contents/MacOS/Arduino\ IDE
/Applications/Arduino IDE.app/Contents/MacOS/Arduino IDE
```

None of these work:

```
"arduino.path": "/Applications
```

```
"arduino.path": "/Applications/Arduino IDE.app",
"arduino.commandPath": "Contents/MacOS/Arduino IDE"
```

```
"arduino.path": "/Applications/Arduino\\ IDE.app",
"arduino.commandPath": "Contents/MacOS/Arduino\\ IDE"
```

Lythinari, 2022-12-07:
The following should work for OSX
```
"arduino.path": "/Applications/Arduino IDE.app/Contents/Resources/app/node_modules/arduino-ide-extension/build/",
"arduino.useArduinoCli": true,
"arduino.commandPath": "arduino-cli"

hujunhan, 2022-12-08:
> The following should work for OSX
> 
> ```
> "arduino.path": "/Applications/Arduino IDE.app/Contents/Resources/app/node_modules/arduino-ide-extension/build/",
> "arduino.useArduinoCli": true,
> "arduino.commandPath": "arduino-cli"
> ```

This works for me.
Tested on a M1 Macbook with Arduino 2

ericoud, 2022-12-18:
> The following should work for OSX
> 
> ```
> "arduino.path": "/Applications/Arduino IDE.app/Contents/Resources/app/node_modules/arduino-ide-extension/build/",
> "arduino.useArduinoCli": true,
> "arduino.commandPath": "arduino-cli"
> ```

It works fine, thank you for sharing !

ScreamZ, 2022-12-22:
> The following should work for OSX
> 
> ```
> "arduino.path": "/Applications/Arduino IDE.app/Contents/Resources/app/node_modules/arduino-ide-extension/build/",
> "arduino.useArduinoCli": true,
> "arduino.commandPath": "arduino-cli"
> ```

Totally hacky, but works :D

craigdetter, 2023-01-16:
Agree, these solutions work. This is totally non-intuitive.

It took me a while to find this article. The exact VSCode error message text was:
Cannot find Arduino IDE. Please specify the "arduino.path" in the User Settings. Requires a restart after change.

Solution was VSCode:
(NOTE: For simplicity of specifying the arduino.path, I renamed [/Applications/Arduino IDE.app] to [/Applications/ArduinoIDE_2.0.3.app]  (removing the space, adding the version number...

The complete solution path was then...

Code -> Settings -> Settings -> type in "arduino.path", then set:

**Arduino: Path**
/Applications/ArduinoIDE_2.0.3.app/Contents/Resources/app/node_modules/arduino-ide-extension/build/

**Arduino: Command Path**
arduino-cli

**Arduino: Use Arduino Cli**
[Checked]