microsoft/vscode-arduino #842: Can't upload or verify - Exit with code=EPERM

assumptionsoup, 2019-07-01:
Using ArduinoIDE_1.8.21.0_x86 on windows with the arduino feather esp32 board.

I can compile and verify using the arduino ide. Vscode seems to have found the ide, because it was logging other errors until I set the arduino.path. 

```
[Starting] Upload sketch - blink.ino
[Warning] Output path is not specified. Unable to reuse previously compiled files. Upload could be slow. See README.
[Error] Exit with code=EPERM
```

LEAFERx, 2019-07-13:
Same problem. It also occurred after output path was set. There was also no build logs (even the build folder was not created).

jkwphysics, 2019-07-21:
Same here, except I do have the output configured.  Any ideas what EPERM means?

jkwphysics, 2019-07-21:
Unable to verify or upload with Arduino Uno R3.  Using Windows 10 app with same result.
"[Error] Exit with code=EPERM"

mMerlin, 2019-07-22:
> Same problem. It also occurred after output path was set. There was also no build logs (even the build folder was not created).

The folder specified with `output` in `arduino.json` needs to exist before attempting a compile.  It is never (and not supposed to be) created.  It is an area to store work in progress and compile history information, but it needs to exist to be used.

That is separate from the other problem.  The initial poster referenced blink.ino though.  If that is from the standard examples, it is possible that the compiler does not have an active writeable folder to work with.  Setting the `output` folder to something reasonable, then looking at that folder contents after failure might provide useful information.  One of the files created there, `build.options.json` shows what the compiler is attempting to do.  If the process gets far enough to create that, it might contain hints about what the problem is.

LuckyXixi, 2019-08-02:
Same Problem. I use Win10 Arduino app with the same result. So I try download the installer from the official website. Then everything seems like work well!

ps: even if you not create the output folder, the arduino plugin well automatically help you create one, but make sure you already have setted up the output path in arduino.json

jkwphysics, 2019-10-31:
Defining an output path in the json just changes the error to "Please see the build logs in Output path:" which is actually still just an empty folder...This still isn't fixed.

hellyzh, 2019-11-01:
Win10 Arduino app from App Store is not supported, please use windows installer version. I use the latest Arduino IDE 1.8.10 and could not repro the issue. Please try and see if the issue still exist.

xinyiz1226, 2019-11-27:
This issue has been marked as stale because it has no recent activities. It will be closed if no further activity occurs within 3 days. Thank you for your contributions.

adiazulay, 2020-11-17:
closed due to inactivity on this issue. Feel free to reopen if this is still a problem.

closed 2020-11-17 by adiazulay

aleksandarzagorac, 2020-11-19:
Yes this is still a problem