microsoft/vscode-arduino #843: Can't verify or compile ESP8266 libs

Nathan-ma, 2019-07-02:
Hello, it seems vscode extension for arduino cannot find basic Libraries like ESP8266WiFi.h

The same code compiles just fine on Arduino IDE, but it cannot even verify properly on vscode.

[image]
I already included the path for my other libraries.

The problem seems to be with the command `#include ''`, this commands tells the compiler to search for files on the directory of the .ino file, thing is, the libs on esp8266 included that way are inside the src folder on the directory of the lib.

Here a few pictures to explain it better:
[image]

Vscode error, not been able to find the BearSSLHealpers.cpp inside the Webserver lib.

[image]
Arduino IDE can find it no problem... path is correct also.

and finally

[image]

so, the .h is inside src on the libs folder. arduino understands it just fine, but vscode try to look for the .h on the rootpath and returns me FILE NO FOUND.

How can I fix this?

Nathan-ma, 2019-07-02:
@pmckinney1971 has fixed this with :
```
"includePath": [
                "C:\\Users\\Leonie\\Documents\\Arduino\\hardware\\esp8266com\\esp8266\\libraries\\ESP8266WiFi\\src",
                "C:\\Users\\Leonie\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\**",
                "C:\\Users\\Leonie\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.0\\**",
                "C:\\Users\\Leonie\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.0\\libraries\\**",
                "C:\\Users\\Leonie\\AppData\\Local\\Arduino15\\**",
                "C:\\Users\\Leonie\\Documents\\Arduino\\hardware\\esp8266com\\esp8266\\**",
                "C:\\Users\\Leonie\\Documents\\Arduino\\libraries\\**",
                "${workspaceFolder}/**"
```
as its includePath, to directly include all the files inse the WiFi/src lib...

ArthurMa1978, 2019-07-15:
thanks @Nathan-ma and @pmckinney1971

closed 2019-07-15 by ArthurMa1978