microsoft/vscode-arduino #847: Can't use baud rate  >=250000

Combinacijus, 2019-07-04:
I have set default baud rate to 1M or chose 250000 and now I get error:
_Error: Inappropriate ioctl for device setting custom baud rate of 250000_
Also it's the same error for 250000 baud rate. And you can't change baud rate until you restart vscode.

TiiFuchs, 2020-11-14:
I need to use the serial monitor with a baud rate of 1000000 (1 million) but there is no way to select that.
Now I always have to use the serial monitor in Arduino IDE. I would be very happy to use it all in one app.

Any fix plannend? This issue is nearly untouched for over a year now.

TiiFuchs, 2020-11-15:
I just edited the files in %userprofile\.vscode\extensions\vsciot-vscode.vscode-arduino-0.3.3\out\src

The baudrate is listed in 
serialmonitor/serialMonitor.js
and in arduino/arduinoSettings.js at the bottom

After adding 1000000 to the array, everything works fine for me...

TiiFuchs, 2021-01-17:
Seems to be done with #1142

closed 2021-01-19 by adiazulay