microsoft/vscode-arduino #851: Unable to start serial monitor

szormok, 2019-07-04:
The serial monitor worked fine but couple of days ago something happened because If I try to open the following error appears:
"Failed to open serial port COM6 due to error: + TypeError: Cannot read property 'close' of null"
Similar as in issue #596.
Also I am not able to change the serial port because the dialog not opens.
I have the latest 1.36.0 and 0.2.26 version.
By the way the uploading works! So the port is alive.
And also in the Arduino IDE everything is fine.

MichaelPfezer, 2019-07-05:
Same issue here

CodeNameHawk, 2019-07-05:
Same issue here

rido-min, 2019-07-05:
I also see this issue

rafalp9728, 2019-07-05:
Also have this issue. It started after update to VS Code 1.36.0.
I see it's been reported already in issue #848.

Cube-Line, 2019-07-06:
+1  like:   ----->>>>     Failed to open serial port COM3 due to error: + TypeError: Cannot read property 'close' of null

ArthurMa1978, 2019-07-08:
We are working on the fix will release new version soon.

ArthurMa1978, 2019-07-09:
Fixed since v0.2.27, thanks @szormok @rido-min @MichaelPfezer @CodeNameHawk @Cube-Line @rafalp9728 .

closed 2019-07-09 by ArthurMa1978

DanwithArduino, 2019-07-09:
trying to change the serial port "crashes the extension host".

CodeNameHawk, 2019-07-11:
@ArthurMa1978 
now work Select Serial Port, and Open Serial Monitor,
but I use an USB-COM adapter FTDI FT232RL with control DTR and RTS, to automatic programming ESP8266
Unfortunately, the plugin incorrectly sets the levels to DTR and RTS, which does not allow to use normally Serial Monitor

ArthurMa1978, 2019-07-15:
@CodeNameHawk thanks for report this, can you help to file a new issue to us?

CodeNameHawk, 2019-07-15:
@ArthurMa1978 Ok.

dzuda11, 2019-11-14:
> We are working on the fix will release new version soon.

Hey!! We have the same issues again with VS Code v1.40.1. can you fix it again or show how can it be fixed? Than you!

xinyiz1226, 2019-11-14:
Duplicate with #918 , we are working on the fix and will let you know when it is ready.

fmlin429, 2020-03-09:
Have the problem resolved? looks like I have the same issue.

dzuda11, 2020-03-09:
> Have the problem resolved? looks like I have the same issue.

Did you update VS Code and extension?
If you are up-to-date then we have the same problem again....

cooperbaker, 2020-03-10:
Same problem here :(

dzuda11, 2020-03-10:
> Same problem here :(

Duplicated with #980 
Downgrade VS code to 1.42 and wait for Arduino extension update....

wouterds, 2020-03-10:
> Downgrade VS code to 1.42 and wait for Arduino extension update....

Can confirm that downgrading to [January 2020 release](https://code.visualstudio.com/updates/v1_42) works ❤️!

DeadpoolKerns, 2020-03-21:
I am having the same issue and would love to downgrade but it appears that the January version is only offering a Win64 release. I downloaded the file VSCodeUserSetup-x64-1.42.1.exe.  Although the download link has a reference of https://update.code.visualstudio.com/1.42.1/win32-x64-user/stable ,which might infer an offer for 32 or 64 bit installations, the download was a Win64 release.

Where can I download the 1.42.1 Win32 release?

333fps, 2020-03-21:
https://az764295.vo.msecnd.net/stable/c47d83b293181d9be64f27ff093689e8e7aed054/VSCodeUserSetup-ia32-1.42.1.exe

DeadpoolKerns, 2020-03-21:
Thank you very much 333fps. I installed 1.42.1 and the serial monitor is functioning as expected again. You've made my day. Now on to getting used to Vscode :+1:  after using arduino UI  :(  for so long

GregTerrell, 2020-03-21:
> I am having the same issue and would love to downgrade but it appears that the January version is only offering a Win64 release. I downloaded the file VSCodeUserSetup-x64-1.42.1.exe. Although the [...]

https://code.visualstudio.com site, then click Updates. This is a history of monthly releases. The link below is the redirect as of today. Left navigation to January 2020 is 1.42.1. Confirmed that rolling back to 1.42.1 fixes the serial monitor fails to open for my Windows10 install.

https://code.visualstudio.com/updates/v1_43

DeadpoolKerns, 2020-03-22:
Thanks Greg but the file I referenced in my comment came from that site. The only January version available for download is the win64 version. This will not install on my Windows 7 32 bit OS

Awbmilne, 2020-03-23:
@fmlin429 @dzuda11 @cooperbaker @wouterds @DeadpoolKerns 

I created issue #1002 for this problem resurfacing, please post a comment there aswell. Thanks!!