microsoft/vscode-arduino #856: An option to use arduino-cli instead of Arduino IDE

TomasHubelbauer, 2019-07-09:
This is a great extension, thanks for making it! I see that it requires the Arduino IDE be installed for one to be able to use it. Arduino recently (about a year ago) started working on `arduino-cli` which is a CLI for Arduino (duh) and I am wondering if you guys have this on your radar and maybe are thinking about using it as a "backend" for this extension instead of the Arduino CLI?

It is still alpha quality but when it's more stable it might make sense. Or maybe it could be offered as an option, an alternative to installing the Arduino IDE?

Some extensions (like the C++ one or the C# one with OmniShrap) already offer to download their prerequisites by themselves so that the user doesn't need to install anything and I am curious if it would make sense for this one to offer to download the Arduino CLI for the user so the only thing to do as a user would be to install this VS Code extension and connect the board, the rest taken care of.

Of course it would probably also be possible to offer automated download of the Arduino CLI but it might be too big or not have a quite mode installer, I haven't looked into that so I don't know.

Will be happy to hear your thoughts on this!

botbench, 2019-07-09:
Came here to leave a similar message, but you beat me to it.

rei-vilo, 2019-08-15:
@ArthurMa1978 

Some references:
+ [Announcing the Arduino Command Line Interface (CLI)](https://blog.arduino.cc/2018/08/24/announcing-the-arduino-command-line-interface-cli/)
+ [Arduino command line interface](https://github.com/arduino/arduino-cli) repository

iqapps, 2019-09-30:
Agree on this

a7hybnj2, 2019-10-11:
there is also the "arduino builder" which appears to have been a similar concept to the "arduino cli" I don't know which is better or what the specific differences are just thought I would mention it just in case...

modmatic, 2019-11-23:
This also has the benefit of the screen not switching to the Arduino IDE every time I verify or upload. I use full-screen apps a lot (Mac) and once it switches over to whichever screen has the IDE, it doesn't go back (that part is an OS thing, but is an annoying side effect of using the Arduino IDE vs CLI).

TomasHubelbauer, 2019-11-25:
@a7hybnj2 It looks like the Arduino Builder is actually based on invoking the Arduino CLI, at least that what I get from a quick read of its source code. It imports it as a module dependency and I think it invokes it for the compilation, although I don't know much about Go so I might be misunderstanding. What's also interesting though is that it is apparently bundled with the Arduino IDE and if it indeed uses the CLI that might mean the CLI is also bundled with the IDE? I have no computer with the IDE at hand right now to check.

If that's true and both the Builder and the CLI come with the IDE, it makes even more sense to offer either as an alternative to the IDE in this extension as it would not present any new prerequisites, people would be required to just have the IDE as they do currently. Still, it would be extra cool if one didn't have to have the IDE if they don't intend to use it and could just have the extension automatically install the CLI by itself, but even not having to install anything extra on top of the IDE is a good argument for adding the CLI option as it won't present new prerequisites aside from the already existing ones.

MarcLV, 2020-09-15:
Guys, please like and comment this PR to fix this once and for all.

orlade, 2021-08-14:
This issue appears to be resolved (i.e. the option exists now), but I haven't managed to get it working since I can't select a board type after enabling CLI (Mac).

a7hybnj2, 2021-11-27:
What is the status of this? Can I install and test this?

I see in the PR that mac and linux support isn't complete. What is still missing?

kopp, 2022-12-30:
@TomasHubelbauer : I can use the plugin with `arduino-cli` at the moment, so I guess this issue can be closed.

TomasHubelbauer, 2023-02-01:
@kopp Did you have to do any setup? I can't test this right now but I'd rather not close until there are clear steps in the readme documenting how to use the CLI over the IDE.

benmcmorran, 2023-02-16:
Thanks for the feedback. [Only the legacy IDE and Arduino CLI are currently officially supported](https://github.com/microsoft/vscode-arduino/blob/374c78f8ed06c1e88df480b687c24c06b728e13b/README.md?plain=1#L39). Additionally, **we plan to drop support for the legacy IDE entirely in the next release of this extension** and recommend that all users move to Arduino CLI as soon as possible. The recent `v0.5.0` release of the extension includes a bundled copy of Arduino CLI that should ease this migration.

kopp, 2023-04-02:
For future reference: I documented the setup I needed to do to use the arduino-cli in #1562

closed 2023-05-09 by github-actions[bot]