microsoft/vscode-arduino #857: Disconnecting Arduino (CH340) while Serial is connected, crashes Extension Host

iqapps, 2019-07-10:
While developing code for Arduino's you often need to disconnect Arduino to test some special functions.

If you disconnect the Arduino while the Serial Monitor has the Serial Port opened, the Visual Studio Code Extension Host crashes and shows the "Restart Extension Host" button.

Could we please have it so that the Serial Monitor just closes the serial port silently when the Arduino is disconnected.

ameer1234567890, 2019-08-20:
happens for me too

gcampbell-msft, 2023-02-16:
@iqapps We have recently updated the Arduino extension to depend on a new standalone Serial Monitor extension, please upgrade your arduino extension and try out this new feature! 

Let us know if it solves your issue!

closed 2023-04-09 by github-actions[bot]