microsoft/vscode-arduino #866: identifier "Serial1" is undefined

clarkbremer, 2019-07-13:
I have an Arduino Mega 2560.  It has 3 serial ports.  The standard is "Serial", the others are "Serial1" and "Serial2".  VSCode doesn't have a problem with "Serial", but shows an error for "Serial1".  Code compiles and uploads just fine, but would be nice not get this error message.  

Version: 1.36.1 (user setup)
Commit: 2213894ea0415ee8c85c5eea0d0ff81ecc191529
Date: 2019-07-08T22:59:35.033Z
Electron: 4.2.5
Chrome: 69.0.3497.128
Node.js: 10.11.0
V8: 6.9.427.31-electron.0
OS: Windows_NT x64 10.0.17134

Name: Arduino
Id: vsciot-vscode.vscode-arduino
Description: Arduino for Visual Studio Code
Version: 0.2.27
Publisher: Microsoft
VS Marketplace Link: https://marketplace.visualstudio.com/items?itemName=vsciot-vscode.vscode-arduino

mMerlin, 2019-10-23:
Try adding
```json
"defines": ["USBCON"]
```
to `"configurations"` in the c_cpp_properties.json file in the .vscode folder.  Or adding `"USBCON"` to the list, if `"defines"` already exists.

jimaobian, 2019-11-20:
If you are using arduino UNO (ATmega328P chip) try:
```
"defines": [
                "__AVR_ATmega328P__",
                "ARDUINO=10800"
           ]
```

If you are using arduino Leonardo (ATmega32U4 chip) try:
```
"defines": [
                "__AVR_ATmega32U4__",
                "ARDUINO=10800"
           ]
```

If you are using arduino Mega (ATmega2560 chip) try:
```
"defines": [
                "__AVR_ATmega2560__",
                "ARDUINO=10800"
           ]
```

Here is the io.h file in AVR library for reference.
```
/* Copyright (c) 2002,2003,2005,2006,2007 Marek Michalkiewicz, Joerg Wunsch
   Copyright (c) 2007 Eric B. Weddington
   All rights reserved.

   Redistribution and use in source and binary forms, with or without
   modification, are permitted provided that the following conditions are met:

   * Redistributions of source code must retain the above copyright
     notice, this list of conditions and the following disclaimer.

   * Redistributions in binary form must reproduce the above copyright
     notice, this list of conditions and the following disclaimer in
     the documentation and/or other materials provided with the
     distribution.

   * Neither the name of the copyright holders nor the names of
     contributors may be used to endorse or promote products derived
     from this software without specific prior written permission.

  THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS"
  AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
  IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE
  ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT OWNER OR CONTRIBUTORS BE
  LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR
  CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF
  SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS
  INTERRUPTION) HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN
  CONTRACT, STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE)
  ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE
  POSSIBILITY OF SUCH DAMAGE. */

/* $Id$ */

/** \file */
/** \defgroup avr_io <avr/io.h>: AVR device-specific IO definitions
    \code #include <avr/io.h> \endcode

    This header file includes the apropriate IO definitions for the
    device that has been specified by the <tt>-mmcu=</tt> compiler
    command-line switch.  This is done by diverting to the appropriate
    file <tt>&lt;avr/io</tt><em>XXXX</em><tt>.h&gt;</tt> which should
    never be included directly.  Some register names common to all
    AVR devices are defined directly within <tt>&lt;avr/common.h&gt;</tt>,
    which is included in <tt>&lt;avr/io.h&gt;</tt>,
[... 652 lines cut ...]
#else
#  if !defined(__COMPILING_AVR_LIBC__)
#    warning "device type not defined"
#  endif
#endif

#include <avr/portpins.h>

#include <avr/common.h>

#include <avr/version.h>

#if __AVR_ARCH__ >= 100
#  include <avr/xmega.h>
#endif

/* Include fuse.h after individual IO header files. */
#include <avr/fuse.h>

/* Include lock.h after individual IO header files. */
#include <avr/lock.h>

#endif /* _AVR_IO_H_ */

```

Xring-git, 2020-03-12:
The same for me, but with:

TCCR2A
TCCR2B
TCNT2
OCR2A
WGM21
CS22
CS20

Code compiles and uploads just fine, but with "Not defined" error.

With F12 appear multiple .h files, seems that "__AVR_ATmega328P__" define is ignored.

thomasio101, 2020-11-08:
# The problem
I had a similar issue, the issue with `Serial` was indeed resolved by adding `"USBCON"` to `defines`, but I still had issues with identifiers such as `DDRD` being undefined. (After writing this answer, I discovered that @jimaobian's answer has the same effect.)
# TLDR
Add the following properties to `c_cpp_properties.json`:
```json
"compilerArgs": ["-mmcu=atmega328"],
"defines": ["USBCON"]
```
# Explanation
After some research, I discovered that the definition for `DDRD` can be found in multiple files (each of which is for a different MCU version). For example, this is a definition from `C:\Program Files (x86)\Arduino\hardware\tools\avr\avr\include\avr\iom328p.h`;
```h
#define DDRD _SFR_IO8(0x0A)
```
At the top of that file, I found this comment;
```h
/* This file should only be included from <avr/io.h>, never directly. */
```
Thus, I took a look at `C:\Program Files (x86)\Arduino\hardware\tools\avr\avr\include\avr\io.h`. That file included the following comment;
> This header file includes the apropriate IO definitions for the
    device that has been specified by the <tt>-mmcu=</tt> compiler
    command-line switch.  This is done by diverting to the appropriate
    file <tt>&lt;avr/io</tt><em>XXXX</em><tt>.h&gt;</tt> which should
    never be included directly.  Some register names common to all
    AVR devices are defined directly within <tt>&lt;avr/common.h&gt;</tt>,
    which is included in <tt>&lt;avr/io.h&gt;</tt>,
    but most of the details come from the respective include file.

After finding this, I managed to find the `-mmcu` argument in [this documentation](https://gcc.gnu.org/onlinedocs/gcc/AVR-Options.html). Based on that documentation I concluded that I need to pass `-mmcu=atmega328` to the compiler for it to work with an Arduino with an ATmega328 MCU.

Now, my `c_cpp_properties.json` looks like this;
```json
{
    "configurations": [
        {
            "name": "Win32",
            "includePath": [
                "${workspaceFolder}/**",
                "C:/Program Files (x86)/Arduino/hardware/arduino/avr/cores/arduino",
                "C:/Program Files (x86)/Arduino/libraries/**",
                "C:/Program Files (x86)/Arduino/hardware/arduino/avr/libraries/EEPROM/src",
                "C:/Program Files (x86)/Arduino/hardware/arduino/avr/libraries/HID/src",
                "C:/Program Files (x86)/Arduino/hardware/arduino/avr/libraries/SoftwareSerial/src",
                "C:/Program Files (x86)/Arduino/hardware/arduino/avr/libraries/SPI/src",
                "C:/Program Files (x86)/Arduino/hardware/arduino/avr/libraries/Wire/src",
                "C:/Program Files (x86)/Arduino/tools/**",
                "C:/Program Files (x86)/Arduino/hardware/arduino/avr/**",
                "C:/Program Files (x86)/Arduino/hardware/tools/avr/avr/include",
                "C:/Program Files (x86)/Arduino/hardware/tools/avr/lib/gcc/avr/7.3.0/include"
            ],
            "forcedInclude": [
                "C:/Program Files (x86)/Arduino/hardware/arduino/avr/cores/arduino/Arduino.h"
            ],
            "compilerArgs": ["-mmcu=atmega328"],
            "intelliSenseMode": "clang-x64",
            "compilerPath": "C:/Program Files (x86)/Arduino/hardware/tools/avr/bin/avr-gcc.exe",
            "cStandard": "c17",
            "cppStandard": "c++17",
            "defines": ["USBCON"]
        }
    ],
    "version": 4
}
```

YakovL, 2020-11-29:
I had problems similar to those of @Xring-git :
`OUTPUT`, `PORTC`,
`TCCR1A`, (but not `TCCR1B`!), `TCNT1`, `OCR1A`, `WGM12`, `CS10`, `TIMSK1`, `OCIE1A`, `TIMER1_COMPA_vect`
all give errors like

> identifier "DDRC" is undefined

which don't prevent sketch from compiling/uploading.

When I opened `./vscode/c_cpp_properties.json` and changed

            "defines": [
                "USBCON"
            ],

to

            "defines": [
                "__AVR_ATmega328P__",
                "ARDUINO=10800",
                "USBCON"
            ],

All those errors went away.

However, I still have errors for `noInterrupts`, `interrupts` and `ISR` functions. They tell specifically:
```
#define noInterrupts() cli()
Expands to:

__asm__ __volatile__ ("cli" ::: "memory")
identifier "__asm__" is undefined C/C++(20)
```
and
```
#define noInterrupts() sei()
Expands to:

__asm__ __volatile__ ("sei" ::: "memory")
identifier "__asm__" is undefined C/C++(20)
```
and
```
#define ISR(vector,...) extern "C" void vector (void) __attribute__ ((signal,__INTR_ATTRS)) __VA_ARGS__; void vector (void)
Expands to:

extern "C" void __vector_11 (void) __attribute__ ((signal,used)) ; void __vector_11 (void)
expected a '{'C/C++ (130)
```
I wonder if there's a more systematic approach to fix these, also without editing c_cpp_properties.json per each sketch.

adiazulay, 2020-11-30:
Thanks everyone for the feedback here! #1141 should fix these issues, and it's very close to being ready to merge. I'll post here again when we have a release candidate that's ready for testing.

adiazulay, 2021-03-22:
IntelliSense auto-generation has been added in v0.4.0. If you're having any problems with IntelliSense please open a new issue.

closed 2021-03-22 by adiazulay

2bam, 2021-08-08:
> If you are using arduino Mega (ATmega2560 chip) try:
> 
> ```
> "defines": [
>                 "__AVR_ATmega2560__",
>                 "ARDUINO=10800"
>            ]
> ```

The `__AVR_ATmega2560__` solved it for me. `USBCON` is **not** a solution, just a patch for people using only `Serial`
I need to access `Serial`, `Serial1 `& `Serial2 `and yours is the only way to make it work.

But I need to patch it every time the intellisense file gets regenerated, it's very annoying.

hlovdal, 2022-02-07:
@2bam Regarding "But I need to patch it every time the intellisense file gets regenerated, it's very annoying.", instead of modifying `c_cpp_properties.json` you can instead add the following

```json
    "buildPreferences": [
        ["build.extra_flags", "-D__AVR_ATmega2560__"]
    ]
```

to `.vscode/arduino.json`. Then those flags will be added on top of what vscode itself generates of content. It is documented in the [readme file](https://github.com/microsoft/vscode-arduino) (not sure how new this is, I just discovered this myself after a long period where I have not worked with arduino. Earlier you could have your own modifications survive in `c_cpp_properties.json`).

valeriangcz, 2022-07-29:
> The `__AVR_ATmega2560__` solved it for me. `USBCON` is **not** a solution, just a patch for people using only `Serial` I need to access `Serial`, `Serial1 `& `Serial2 `and yours is the only way to [...]

Hi, I find a solution to avoid patch it everytime, if you use vscode.
1. open settings
2. search "arduino"
3. find options  "Arduino: Disable Intelli Sense Auto Gen"
4. check this options, disable the "intell sense Auto Gen"